SystemRDL/PeakRDL-regblock #2: Frontend command line application

PeterCDMcLean, 2022-03-24:
I believe that this tool could really use a frontend command line application.
Example:

```
python peakrdl_regblock.py top.rdl submodule1.rdl submodule2.rdl --root=top --output_dir=/path/to/output_dir --cpu_if=APB3_Cpuif
```

amykyta3, 2022-03-25:
I completely agree and that is the plan!
The command line application will be implemented here as a unified tool: https://github.com/SystemRDL/PeakRDL
It will not only include this regblock generator, but also the other exporter tools in this github project space.

The reason I'm publishing these components separately is because it gives users the flexibility to build their own custom wrapper toolkits to suit their specific needs.

amykyta3, 2022-07-30:
Command line application is now available. See: https://peakrdl.readthedocs.io/en/latest/

closed 2022-07-30 by amykyta3