SystemRDL/PeakRDL-regblock #4: Add support for external

PeterCDMcLean, 2022-03-30:
As is mentioned in the known limitations, PeakRDL-regblock has no support for external registers or regfile. Having support for external is necessary for certain common use cases:

 - FIFO read access
 - Mapping to block RAM read
 - Any other read that must be qualified with a 'read enable' and has arbitrary latency

My suggestion is that the 'out' struct get a scoped 'read enable' signal for any external read. The 'in' struct will contain a 'valid' signal indicating the read is complete and the 'in' data is to be forwarded back to the axi/ahb/cpu bus

amykyta3, 2022-03-31:
Yep thats essentially the plan.

For external `addrmap` and `regfile` child nodes, the `hwif_in`/`hwif_out` structs will contain handshake signals that follow the internal CPUIF protocol described here: https://peakrdl-regblock.readthedocs.io/en/latest/cpuif/internal_protocol.html

External `reg` and `field` nodes will have a simpler handshake that allows for arbitrary access latency. Probably pretty similar to the internal protocol, but just without the address bus.

amykyta3, 2023-05-12:
Published in `v0.13.0`

closed 2023-05-12 by amykyta3