SystemRDL/PeakRDL-regblock #7: Vivado Simulator Issue

mkahane, 2022-05-02:
This is almost certainly an issue with XIlinx, but I wanted to point out that I am seeing a Vivado Simulator Failure when running a simulation with a generated register block:

FATAL_ERROR: Vivado Simulator kernel has discovered an exceptional condition from which it cannot recover. Process will terminate. For technical support on this issue, please open a WebCase with this project attached at http://www.xilinx.com/support.
Time: 0 ps  Iteration: 0  

It does give the line of RTL where it fails in the comb block where s_axil.RDATA is assgined:

https://github.com/SystemRDL/PeakRDL-regblock/blob/main/src/peakrdl/regblock/cpuif/axi4lite/axi4lite_tmpl.sv#L208

which generates the line:

`  s_axil.RDATA = axil_resp_buffer[axil_resp_rptr[1:0]].rdata;
`

I was able to hide the error by making the axil_resp_buffer a power of 2 depth, but the waveform didn't indicate any other issues.

amykyta3, 2022-05-03:
Thanks for pointing it out. Yeah i wonder if Vivado's simulator is panicking when the index is out of range. I'll rework that line to make it slightly less sloppy and handle the non-power of two case better despite it being logically impossible.

The Vivado simulator is basically worthless - I tried using it for functional simulations for unit tests and it [fell apart any time I tried to do anything remotely fancy](https://github.com/SystemRDL/PeakRDL-regblock/blob/main/tests/lib/simulators/xilinx.py#L9-L13) on the testbench side of things.
I have instructions [here on how to get the Questa simulator for free](https://github.com/SystemRDL/PeakRDL-regblock/blob/main/tests/README.md).

I'll scope in adding a sanity compilation check at least since I'm sure people use it for general one-off TBs.

amykyta3, 2022-05-03:
Reproduced the issue on my end.
I re-enabled the Vivado simulator to run on some of the testcases to catch tool quirks like this in the future.

Also added your workaround for now since it is definitely the simplest and ends up getting optimized away anyways,
Thanks!

closed 2022-05-03 by amykyta3