SystemRDL/PeakRDL-regblock #20: Sample code license

jeremiah, 2022-09-27:
Hi,

Is the sample code in your documentation licensed? It would be good if it were explicitly licensed so people would know what rights and obligations they have should they choose to reuse the sample code. 

Thanks

amykyta3, 2022-09-28:
Any examples in the documentation are unlicensed and free to use however you see fit. Since they are examples intended to demonstrate how anyone can use the library, it is pretty implicit that they will be freely copied by anyone intending to learn how to use this tool.

The GPL v3 license that the PeakRDL projects are covered by are only intended to cover the respective code generator tool itself. (generally, anything that resides within the [src/](https://github.com/SystemRDL/PeakRDL-regblock/tree/main/src) folder)
This means that if you integrate the code generator into a larger project, the obligations outlined in the [GPL v3 license](https://github.com/SystemRDL/PeakRDL-regblock/blob/main/LICENSE) apply.

I should note - any content that is *generated* by the tool is free to use any way you see fit, as it is built off of your custom input RDL definitions. For example if you generate a SystemVerilog regblock using this tool, and it is integrated into your ASIC/FPGA project, that SystemVerilog code is entirely yours as it falls outside of the scope of the license.

This is a pretty common question I get. I'll work on posting some official clarification in the docs.
Hope that explanation helps.

jeremiah, 2022-09-28:
Thanks for the reply. Unfortunately, implicit license terms are often not suitable for commercial projects, so having your explicit statements are valuable - thanks!

Thanks for the explanation about the GPL'd tool's output. Your description is similar to the way the GNU Compiler Collection uses GPLv3 as well although they use an exception to make explicit what the output may be licensed; https://www.gnu.org/licenses/gcc-exception-3.1.en.html This might be an avenue you chose should you wish to. Please note that I say this because I respect your license choice and want to comply with the terms you've chosen. 

Thanks again for writing a useful RDL tool and for explaining license usage -- very helpful!

amykyta3, 2022-11-10:
Added some clarification text to the documentation: https://peakrdl-regblock.readthedocs.io/en/latest/licensing.html
Let me know if I missed something.

closed 2022-11-10 by amykyta3