SystemRDL/PeakRDL-regblock #25: Address with appears to max out at 24 bits

asicguy, 2023-01-18:
Hi Alex, hope things are going well.
I'm hoping to move to using systemrdl for my register definitions. I am running into a problem with the address widths. If I understand correctly, peakrdl- regblock is supposed to extract the necessary address width from the register definition. If I use the following rdl code:
addrmap BOEEye_reg {
    default regwidth=64;
    reg {
        field {
            desc   = "FL2 Value";
            sw     = r;
            hw     = w;
        } num[7:0];
    } fl2 @32'h01C00018;
};
The code generated limits the addresses to 25 bits. The problem is that this may cause overlaps with other generated blocks if I am using the whole 32 bit address space. For example, 1C00018 would also decode register accesses to 3C00018 since it is ignoring bit [31:25] being a 0. Is there a way to specify the address width to get around this?

Sorry if this is a stupid question, I'm just starting to explore this.
Thanks,
Frank

amykyta3, 2023-01-19:
Hey Frank!

Not a stupid question at all. You'll notice that in SystemRDL there really isn't any mechanism to reserve space or add address padding - an RDL node's size is always dictated by the highest address offset of its contents. For this reason, the default behavior of the regblock generator is to only use the minimum amount of bits needed to address the range of the block.
Usually the interconnect that drives the block's CPU interface would be sized to match the range of your register block. This would prevent the aliasing you describe. That said, I can see how it would be useful to override it and expand the range.

I just added a feature in v0.10.0 that lets you override the address width.

asicguy, 2023-01-19:
Thanks a bunch. Will give it a try shortly.

closed 2023-01-20 by amykyta3