SystemRDL/PeakRDL-regblock #26: common trigger for array of write-buffered registers

mkahane, 2023-02-07:
Hi team,
Let's say I have a regfile like this:

```
regfile my_file {
  default accesswidth = 32;
  default regwidth = 32;

  reg {
    field {sw =rw; hw=r} trigger_bit;
  } write_trigger;

  reg {
    regwidth = 64;
    field{ sw=rw; hw=r} wide_field[63:0];
    buffer_writes=true;
  } buffered_regs[8];

};
```

My question is this:
1.) how can I concisely apply the write_trigger to all the buffered regs? The example [here](https://peakrdl-regblock.readthedocs.io/en/latest/udps/write_buffering.html#atomic-group-of-registers) seems to indicate that all triggers need to be set individually

2.) This doc seems to indicate that a single bit field can act as the trigger.  What would be the syntax for this? something like 

`buffered_regs[0] -> wbuffer_trigger = write_trigger.trigger_bit  `

is such dereferencing supported?

amykyta3, 2023-02-08:
Just omit the array suffix and it will assign to all elements of the array:
```
buffered_regs->wbuffer_trigger = write_trigger.trigger_bit;
```

This is actually the only supported way at the moment. The compiler [doesn't support assigning individual array elements](https://systemrdl-compiler.readthedocs.io/en/stable/known_issues.html#no-support-for-heterogeneous-arrays) yet.

mkahane, 2023-02-08:
roger. Thanks, Alex!

closed 2023-02-08 by mkahane