SystemRDL/PeakRDL-regblock #34: Consider:

galaviel, 2023-04-05:
PeakRDL seems to emit synchronous reset currently.
Add the option (via cmdline, config file, etc.) to emit Async reset.
Out standard cell libraries uses ASYNC reset for all FF's.

galaviel, 2023-04-21:
Also consider: Reset Polarity.
Currently regblock seems to default to "active high" however I believe most ASIC designs use "active low".
Simple textual search & replace can be used to work-around that; however would be nice to have the toolchain emit suitable polarity in the first place.

amykyta3, 2023-04-22:
Once internal to an ASIC (or FPGA), active high vs active low becomes a matter of preference and generally doesn't matter with most technology nodes I have worked with.
That said, I recognize that organizations will have differing conventions, so this seems like a sensible thing to add.

I'll be sure to add this as a command-line flag as well as a `peakrdl.toml` configuration option, since both could be useful.

In the meantime, be aware that you can already control this using a `cpuif_reset` signal declaration in the top `addrmap`:
```
addrmap top {
    signal {
        cpuif_reset;
        async;
        activelow;
    } rst_n;
};

galaviel, 2023-04-23:
thanks! that sounds great.

Regarding reset, the above RDL code snippet only seems to reset the CPU I/F.

The storage seems to remains reset by 'rst' (active high, sync).

So the feature of controlling the storage's reset is still valuable..
also it's desirable IMHO to be able to separate the access logic from the storage reset.
Sometimes you want to perform 'logic' reset to a unit (including all the bus elements) but retain the configuration (storage) so as to save SW from needing to re-configure the block.

amykyta3, 2023-04-25:
Ah I forgot to add the `field_reset` property too:
```
signal {
    cpuif_reset;
    field_reset;
    async;
    activelow;
} arst_n;
```

But either way, agree this would be a useful thing to add. This will be up next after I finish #4

galaviel, 2023-04-27:
sounds great! trying... (simulation still running..)
but looks good.

amykyta3, 2023-05-14:
Implemented this feature. Will be in the next release.

closed 2023-05-14 by amykyta3

galaviel, 2023-05-14:
Awsome !! thank you!