SystemRDL/PeakRDL-regblock #36: External memory connections

davidp135, 2023-04-11:
Hi, I am just looking at your tool for integrating into the workflow at my company. It looks really good!

One thing that is currently a blocker to using this is the lack of support for external memory connections for the purpose of mapping block RAM to an address map. It looks like you were looking into it in issue https://github.com/SystemRDL/PeakRDL-regblock/issues/4. Are there any plans to role support soon?

I see @ycyang0508  may have been working on this on a fork https://github.com/ycyang0508/PeakRDL-regblock/commit/358a09809514ad385cd1dcef99358699b8c56231 through to https://github.com/ycyang0508/PeakRDL-regblock/commit/381724e743d52d09d2c104f480aa2ff7e4b2627a 

Thanks

David

philipaxer, 2023-04-11:
Hi, 
I was always wondering what the applications of such an sram mapping would be. Could you give me a high level example?

Regards 
Philip

ycyang0508, 2023-04-12:
Hi,
    In my option, HW core can write/read the thousands result into/from SRAM , after that, FW or can read the sram to get results directly. Thus: 
  First, Compare the registers, SRAM is more compact in area resource and read/write interface of sram is more simple. 
  Second, it also simplify design of programming registers design to avoid create a lot of registers with same purposes. 

Ya-Chau

amykyta3, 2023-04-12:
Yes this feature is definitely planned. (Marking this as duplicate of #4)

For now, I would recommend keeping RAM-based regions separate from register files. This is pretty straightforward to do by simply providing a separate port from the interconnect that drives your interfaces.

If there are any specific requests for the functionality of this feature, please comment on issue #4 so I can keep track of any specifics. Otherwise I'm closing this as duplicate.

davidp135, 2023-04-12:
Sure, it is for whenever the host needs access to large data sets for read/write to where storing the data in discrete registers would use up a lot of space inside an FPGA.  For example, say that a peripheral needs to know the xyz coordinates of a large number of objects, a RAM implementation of storage will be optimal. 

There are two ways (that I can think of) to access RAM blocks through an address map:
1. Have 2 addresses in your address map, one for the chosen RAM address and one that is populated with data for that particular RAM. Maybe you would want a write enable too.
2. Have a virtual address for each RAM address. So if your RAM block is 256 words deep, there would be 256 addresses assigned to that RAM. 

From a SW perspective it's more simple to just go to the RAM address and read/modify as shown in option 2. A read or write is a 1 step operation rather than 2. You can just pass around the pointer to where that data is located. A language like C requires a contiguous memory map for it to work correctly.

From what I understand from my initial investigation the `mem` function gives access to this functionality:
https://peakrdl.readthedocs.io/en/latest/systemrdl-tutorial.html

Does this make sense and is there anything that I have missed?

Thanks

David

davidp135, 2023-04-12:
Thanks @amykyta3, I will have a think about doing the manual option.

philipaxer, 2023-04-12:
Thanks a lot @davidp135. does that intrinsically imply dual ported memory? One port for SW and one port for HW?

For this application I would have probably given the IP an AXI master port to fetch data from system main memory via an DMA descriptor. That way you can keep the peripheral memory map smaller.
But I can now see the general idea.

rakesh2215, 2023-04-20:
> Hi, In my option, HW core can write/read the thousands result into/from SRAM , after that, FW or can read the sram to get results directly. Thus: First, Compare the registers, SRAM is more compact [...]

I tried your fork, latest main; that also gave same error.
"Exporter does not support external components"
Is that expected?

ycyang0508, 2023-04-21:
This fork is only for my personal usage before and it's quite long time ago for me to trace those code.

rakesh2215, 2023-04-21:
> 

Sure. I saw a commit message "add external memory support" in your fork; so I gave it a try. 
Thanks for your reply.

@amykyta3 seems this feature is in high demand :)

amykyta3, 2023-04-22:
It sure seems so!
I am humbled by everyone's enthusiasm here.  I have already dusted off my old notes about this topic and am looking into this as the next feature I implement.

rakesh2215, 2023-05-10:
> From what I understand from my initial investigation the `mem` function gives access to this functionality:

@davidp135 ; I am not able to find an example with "mem" keyword. 
Any pointers?
In addition, please suggest if you think "mem" can serve the same purpose as "external".

rakesh2215, 2023-05-10:
I found some more info on "mem" in SystemRDL reference manual.
Seems there is no way around having "external" keyword support.

amykyta3, 2023-05-11:
`mem` are distinct components, like `addrmap` and `regfile`. The difference is that `mem` components are implicitly always external.

BTW: I am wrapping up the changes for external support. Writing up some testcases, flushing out bugs, and updating documentation. The next few weeks are pretty busy for me, but I'm doing what I can!

amykyta3, 2023-05-12:
Published in `v0.13.0`