SystemRDL/PeakRDL-regblock #37: Add support for write broadcasting

amykyta3, 2023-04-12:
A common design pattern is to implement an address region that broadcasts write operations to multiple registers. This saves on software cycles since a single SW write can be used to broadcast a configuration to multiple registers simultaneously.

Since this is not supported by SystemRDL's built-in properties, add this as a UDP extension.

For example:
```systemrdl
my_reg_t rega;
my_reg_t regb;
my_reg_t regc[4];

my_reg_t reg_all;

reg_all -> broadcast_write = '{
    rega, regb, regc
};
```

Writing to `reg_all` is equivalent to individually writing to `rega`, `regb`, and all of `regc`

Semantics:
    - The `broadcast_write` property is valid on `reg` or `regfile` components
    - If a component is assigned `broadcast_write`, its hw implementation is omitted. Instead, all the broadcast targets the property references inherit an additional SW write-enable strobe.
    - `broadcast_write` accepts an array of one or more references.
    - Each reference shall be the exact same component type and datatype as the assignee
    - A reference to a `regfile` applies recursively to all its child registers
    - A reference to an entire array applies to all elements in the array
    - A reference to a specific array element is not supported
    - A broadcaster and its targets shall be internal to each other. (one cannot be external compared to the other)
    - A  software read operation on the broadcaster is illegal and the implementation may optionally return a cpuif error.

amykyta3, 2025-11-20:
I need to dig up my notes on this. I put this idea together several years ago, but it got de-prioritized. I believe I was still undecided on a few architectural things. If its ok, I'll convert your PR to a draft so that it can be used as a reference for a final implementation. Agree there are still several edge-cases that are unsolved.

sbaillou, 2025-11-20:
Yes, that's fine, I know this isn't the optimal way of implementing it. But thanks for looking into it!

I'll try to use my PR locally in the meantime, and revert the changes once we have a cleaner implementation. :)

Do you have a very rough timeline in mind yet? (in a couple of months, or rather a couple of years?)