SystemRDL/PeakRDL-regblock #39: Buffered write, trigger is a field of the register

Risto97, 2023-04-29:
If a field is a trigger for the register like this:
```
property buffer_writes {
    component = reg;
    type = boolean;
};

property wbuffer_trigger {
    component = reg;
    type = ref;
};

addrmap testmap {
    reg {
        regwidth = 32;
        field { fieldwidth = 1; sw = w; hw=r;} commit = 0;
        field { fieldwidth = 8; sw = w; hw=r;} some_field = 0;
        buffer_writes = true;
        wbuffer_trigger = commit;
    } some_reg;
};
```

The problem is in order for commit to update the value of commit in field_storage needs to change, but it will only change once it is commited, right?
So I think it should have this change:
```diff
    always_comb begin
        automatic logic [0:0] next_c = field_storage.some_reg.commit.value;
        automatic logic load_next_c = '0;
-        if(wbuf_storage.some_reg.pending && field_storage.some_reg.commit.value) begin // SW write
+        if(wbuf_storage.some_reg.pending && wbuf_storage.some_reg.data[0:0]) begin // SW write
            next_c = (field_storage.some_reg.commit.value & ~wbuf_storage.some_reg.biten[0:0]) | (wbuf_storage.some_reg.data[0:0] & wbuf_storage.some_reg.biten[0:0]);
            load_next_c = '1;
        end
        field_combo.some_reg.commit.next = next_c;
        field_combo.some_reg.commit.load_next = load_next_c;
    end
```
I think its fine to have trigger in the same register, as it would be possible to strobe (#38) parts of the data and then on strobing the part where commit is located, the data should be commited. 

If you point me where I should look for the fix, I can do it also.

amykyta3, 2023-05-05:
No I disagree.
For write buffering, the value in the buffer is transient and is not intended to represent the current state of the register. Setting `wbuffer_trigger` to a field reference is referring to the value of the storage element defined by the field, not the data that is pending to be committed to it.

I don't think having the trigger being a sub-field of the buffered register makes particular sense here.

Risto97, 2023-05-05:
I see, I do agree that it does not make too much sense, I just run into this scenario while trying to figure out how wbuffer works.
However, I think maybe there should be a compile error when the trigger is a field within the same register?

amykyta3, 2023-05-12:
Agree. Yeah I can add a validation check for that.

closed 2023-10-26 by amykyta3