SystemRDL/PeakRDL-regblock #42: swmod late?

xachb, 2023-06-02:
Using peakrdl-regblock 0.14.0 and I am trying to generate a register that will signal to the hardware when a new value is written.  I believe the swmod property is the best way to go, but I am finding I would have to add an additional flop to use the swmod signal.  It is not clear to me in the SystemRDL 2.0 spec when swmod should be asserted, but I believe it would be more useful if it were 1 cycle later.  

Thank you,
Zach

amykyta3, 2023-06-02:
Although true that the spec does not specify this, this is an intentional design decision to allow the most flexibility. See #19 for more details.

amykyta3, 2023-06-05:
No need to apologize! Glad to hear it is working well. I am always open to feedback on how to improve the tools further.

closed 2023-06-08 by amykyta3