SystemRDL/PeakRDL-regblock #44: Is external memory/register supported ?

DerekWangScaleflux, 2023-06-09:
Hi,
   Is external memory/register supported ?

   I have tried the following:
   external my_reg  ext_my_reg  ;

   But the tool report error:
   error: Exporter does not support external components

Best Regards
Derek Wang

rakesh2215, 2023-06-09:
I was hitting the same error but realized I had the older codebase.

Then on a new machine I tried below, it worked:

python3 -m pip install peakrdl-regblock
python3 -m pip install peakrdl

 peakrdl regblock accelera-generic_example.rdl -o . --cpuif axi4-lite

accelera-generic_example.rdl has "external" registers.

DerekWangScaleflux, 2023-06-09:
> 

Thanks.

Could you post your peakrdl version ?
And how you define the "external" registers in you rdl ?

Best Regards
Derek Wang

rakesh2215, 2023-06-09:
I am at 0.14 version, which had external support added.
Have a look at the example file:
https://github.com/SystemRDL/systemrdl-compiler/blob/main/examples/accelera-generic_example.rdl

amykyta3, 2023-06-09:
Correct. This feature was added in v0.13.0. I always post detailed release notes here: https://github.com/SystemRDL/PeakRDL-regblock/releases

For more details on how external components work, see the documentation: https://peakrdl-regblock.readthedocs.io/en/latest/rdl_features/external.html

DerekWangScaleflux, 2023-06-13:
> Correct. This feature was added in v0.13.0. I always post detailed release notes here: https://github.com/SystemRDL/PeakRDL-regblock/releases
> 
> For more details on how external components work, see the documentation: https://peakrdl-regblock.readthedocs.io/en/latest/rdl_features/external.html

Thanks.
It got solved on the latest version.

closed 2023-06-13 by DerekWangScaleflux