SystemRDL/PeakRDL-regblock #47: Feature Request: Option to remove buffer stage from AXI read/write

leolitenstorrent, 2023-06-12:
Hello PeakRDL Development Team,

I and my project team noticed that there is an additional buffer stage added to the read/write transactions of the AXI Bus. Would it be possible if you add an exporter option to remove the buffer stage, so the read/writes would take only two cycles instead of three?

Thank you very much!

amykyta3, 2023-06-13:
Are you using the latest version? I had a bug in a past version where an extra retiming register stage was accidentally enabled by default.

I did a quick check and I'm pretty sure the latency for AXI4-lite is 2 cycles if all the retiming features are disabled.

amykyta3, 2023-06-14:
Yes. Without any retime stages, 2 clocks is what is expected.

leolitenstorrent, 2023-06-14:
Hello Alex,

Would it be possible that there be an option to reduce the transaction to one clock?

According to the AXI standard, a one-cycle transaction is possible and will be very useful when the digital design will be operating at very high clock frequencies or when low latency is important (i.e. a memory/serializer-deserializer module), which is my team’s case.

We have been using PeakRDL for a long time and really appreciate the library. This way, we can incorporate even more of PeakRDL into our design. We appreciate your consideration.

Thank you!

amykyta3, 2023-06-15:
Yeah I can see if I can tighten up the response time. I agree that if all optional retiming stages are removed, it should result in the minimum latency for a given protocol.

leolitenstorrent, 2023-06-15:
Hello Alex,

Thank you!