SystemRDL/PeakRDL-regblock #48: How to create CSR's software access property controlled by hardware ?

DerekWangScaleflux, 2023-06-13:
Hi,
      How to create a register which SW read and write access property is controlled by hardware ?

      I know that "swwe" and "swwel" is used to control the write access property. But how about read access property ?

     My intentiion is to create a register which could only be accessed when another register is not set.

Best Regards
Derek Wang

amykyta3, 2023-06-13:
I am not aware of a mechanism using SystemRDL's built-in properties that would allow you to describe this behavior.

I would suggest this feature here: https://forums.accellera.org/forum/52-systemrdl-discussion/
Hopefully Accellera can add something to the official spec in the next revision.

Alternatively, SystemRDL's User Defined Properties (UDPs) could be used to describe this construct, and support added to this code generator.

DerekWangScaleflux, 2023-06-13:
> I am not aware of a mechanism using SystemRDL's built-in properties that would allow you to describe this behavior.
> 
> I would suggest requesting this feature here: https://forums.accellera.org/forum/52-systemrdl-discussion/ Hopefully Accellera can add something to the official spec in the next revision.
> 
> Alternatively, SystemRDL's User Defined Properties (UDPs) could be used to describe this construct, and support added to this code generator.

Thanks. 
I think external register could be used for this purpose.

closed 2023-06-15 by amykyta3