SystemRDL/PeakRDL-regblock #49: Making bit widths of vectors the same when doing comparisons

leolitenstorrent, 2023-06-13:
Hello PeakRDL Development Team,

Is it possible that, in the future versions, the bit widths of vectors be set to the same in comparisons? For example, `cpuif_req_masked & (cpuif_addr == 'h0)` can become `cpuif_req_masked & (cpuif_addr == 8'h0)` when the address width is 8. I know there are other parts of the script that generate these comparisons, and I appreciate that you improve these as well.

I understand this may be a little bit of laborious, but it will stop warning messages about bit widths not matching when I am doing Verilog simulations.

Thank you very much!

amykyta3, 2023-06-15:
I have adjusted the code generator to generate sized integers in nearly all cases I could find.

I'm curious - can you share which simulator you are using that generates these warnings?

leolitenstorrent, 2023-06-15:
Hello Alex,

I am using Synopsys's VCS with lint options enabled such that warnings like unmatched bit-widths, permanently enabled flops, etc. are reported.

amykyta3, 2023-06-29:
Fixed in v0.16.0

closed 2023-06-29 by amykyta3