SystemRDL/PeakRDL-regblock #52: Instantiation stub to use generated RTL

arhum-alam, 2023-06-27:
It would be very helpful to have an option in the tool to generate an instantiation stub that can be used within the design using the register map code to instantiate it. This would be beneficial especially for RDL files that have a lot of fields and would avoid having to write down each field name manually and would also reduce the chance of human error in instantiating the register map code.

This feature was added to our local copy of the PeakRDL code as a new class within the generators.py file in the hardware interface (hwif) directory.

amykyta3, 2023-06-29:
I have already implemented a feature for a similar reason which generates a "hardware interface report".
Have you tried that? (See the `peakrdl regblock --hwif-report` flag)
It dumps a text file of the flattened hwif structure so that it can be easily used during development.

amykyta3, 2023-07-20:
Closing due to inactivity

closed 2023-07-20 by amykyta3