SystemRDL/PeakRDL-regblock #53: Divide by zero for nested addrmaps

ion-concepts, 2023-06-29:
I ran into a divide by zero (% 0 really) when trying to compile a regfile that was defined in multiple files as nested addrmaps. In simplifying the test case I can reduce it to a single simple .rdl file that demonstrates the issue.
Versions: peakrdl-0.8.0 peakrdl-regblock-0.15.0 
```
$ peakrdl regblock test.rdl -o output
Traceback (most recent call last):
  File "/home/ianb/.local/bin/peakrdl", line 8, in <module>
    sys.exit(main())
  File "/home/ianb/.local/lib/python3.8/site-packages/peakrdl/main.py", line 165, in main
    options.subcommand.main(importers, options)
  File "/home/ianb/.local/lib/python3.8/site-packages/peakrdl/subcommand.py", line 154, in main
    self.do_export(top, options)
  File "/home/ianb/.local/lib/python3.8/site-packages/peakrdl_regblock/__peakrdl__.py", line 207, in do_export
    x.export(
  File "/home/ianb/.local/lib/python3.8/site-packages/peakrdl_regblock/exporter.py", line 160, in export
    DesignValidator(self).do_validate()
  File "/home/ianb/.local/lib/python3.8/site-packages/peakrdl_regblock/validate_design.py", line 32, in do_validate
    RDLWalker().walk(self.top_node, self)
  File "/home/ianb/.local/lib/python3.8/site-packages/systemrdl/walker.py", line 150, in walk
    self.current_action = self.do_enter(node, listener)
  File "/home/ianb/.local/lib/python3.8/site-packages/systemrdl/walker.py", line 182, in do_enter
    new_action = listener.enter_AddressableComponent(node) or WalkerAction.Continue
  File "/home/ianb/.local/lib/python3.8/site-packages/peakrdl_regblock/validate_design.py", line 73, in enter_AddressableComponent
    if (node.raw_address_offset % alignment) != 0:
ZeroDivisionError: integer division or modulo by zero
```
This `.rdl` fails: 
```
addrmap test {
    addrmap {
       reg {name="Reg A";
            field {
                  sw=rw;
                  hw=r;
                  name="";
                  desc="";
            } A[31:0] = 0x0;
        } A_reg @ 0x0;
        reg {name="Reg B";
            field {
                  sw=rw;
                  hw=r;
                  name="";
                  desc="";
            } B [31:0] = 0x0;
        } B_reg @ 0x4;
    } mapA @ 0x1000;
};
```
But this `.rdl` works: 
```
addrmap test {
    reg {name="Reg C";
            field {
                  sw=rw;
                  hw=r;
                  name="";
                  desc="";
            } C[31:0] = 0x0;
        } C_reg @ 0x0;
    addrmap {
       reg {name="Reg A";
            field {
                  sw=rw;
                  hw=r;
                  name="";
                  desc="";
            } A[31:0] = 0x0;
        } A_reg @ 0x0;
        reg {name="Reg B";
            field {
                  sw=rw;
                  hw=r;
                  name="";
                  desc="";
            } B [31:0] = 0x0;
        } B_reg @ 0x4;
    } mapA @ 0x1000;
};
```

amykyta3, 2023-06-29:
Interesting edge case. Thanks!

In SystemRDL, `addrmap` components are considered external. This means that in your failing example, the exporter was unable to reasonably guess the bit-width of the CPU interface. For now I made a change that will assume 32-bits for this edge case.

If you want to describe an internal hierarchy within an exported register block, I'd recommend using `regfile` components instead of `addrmap`.

amykyta3, 2023-06-29:
Fixed in v0.16.0

closed 2023-06-29 by amykyta3

amykyta3, 2023-06-30:
Ah of course!
I also use the pattern of nested addmaps all the time. If that was your original intent in your example, then I would recommend placing exportable regblock definitions outside of the lexical scope of your top-level block. For example:

```systemrdl
addrmap peripheral_a {
   reg {
        field {
              sw=rw; hw=r;
        } A[31:0] = 0x0;
    } A_reg @ 0x0;
    reg {
        field {
              sw=rw; hw=r;
        } B [31:0] = 0x0;
    } B_reg @ 0x4;
    // ... etc ...
};

addrmap peripheral_b {
    // ... etc ...
};

addrmap top {
    peripheral_a mapA @ 0x0000;
    peripheral_b mapB @ 0x1000;
};
```

In this example, `top` represents your main interconnect/decoder that connects to one or more peripheral register blocks.
By making the peripherals named declarations in the root namespace, you can then generate their regblock implementations directly:
```
peakrdl regblock example.rdl --top peripheral_a -o output/
peakrdl regblock example.rdl --top peripheral_b -o output/
```
You can also split these apart into separate files which can help organize large designs.
For example:
```bash
# Generate the sub-blocks separately
peakrdl regblock map_a.rdl --top peripheral_a -o output/
peakrdl regblock map_b.rdl --top peripheral_b -o output/

# stitch together a full chip HTML document
peakrdl html map_a.rdl map_b.rdl top.rdl
```