SystemRDL/PeakRDL-regblock #55: Use literals of the same size in comparisons

leolitenstorrent, 2023-07-18:
Hello Alex,

May you please use literals of the same size in comparisons (see image below)? This has been causing Spyglass Lint to generate warning messages. Thank you very much!

[image]

amykyta3, 2023-07-19:
This is not a very practical rule to adhere to. There are numerous cases where comparisons of differing sized vectors is possible, and perfectly appropriate. 

Do you see an issue in the behavior of the logic? According to the SystemVerilog LRM, comparisons are implicitly padded up anyways, so doing so in the generated code is unnecessary:
[image]

leolitenstorrent, 2023-07-19:
Hello Alex,

I understand that literals of different sizes is important in comparisons like this (such as to prevent overflow) and adheres to the SystemVerilog standard.

I wish to make it clear that our analysis tool (Synopsys Spyglass with Lint enabled) is not flagging the unmatched literal sizes as invalid syntax/inappropriate, but rather generating warning messages that serve as reminders that this may be the result of an error/bug (such as when the designer accidentally declares a wire as a bit wider than another and trying to compare the two wires).

Since Synopsys Spyglass is a widely-used analysis tool in industry and these warning messages help us in finding potential errors and bugs in our SystemVerilog code (and we wish to make sure that the code does not generate these warnings when they are not bugs), may I kindly request that PeakRDL generate literals of the same size in comparisons?

Thank you very much!

amykyta3, 2023-07-20:
In this particular case I think it would be inappropriate to modify the implementation to appease the lint tool. SystemRDL's semantic rules already include checks that prevent generation of code that the Spyglass rule is intending to cover. (there are existing RDL property reference width bounds checks for exactly this reason).
In your case, I would recommend waiving the violation in Spyglass.

Modifying the output here would only hinder its readability, and add unnecessary complexity to this code generator.

closed 2023-07-20 by amykyta3