SystemRDL/PeakRDL-regblock #58: Omit unecessary signals if an external register is read-only or write-only

amykyta3, 2023-08-03:
In some cases (as discussed in #57), external registers may be specified as read-only or write-only.
In this situation a read or write access to such a register is not ever expected so there is no reason to include signals for these transfers on the hwif just for the designer has to tie them off.

Update exporter behavior to omit the `wr_data`, `wr_biten`, `wr_ack`, `rd_ack`, and `rd_data` signals as appropriate based on the access policy of the register.

amykyta3, 2023-08-03:
Implemented. Will be included in upcoming release.

closed 2023-08-03 by amykyta3

roowatt, 2023-09-07:
I am still seeing the write interface signals being included despite updating to the latest release (peakrdl-regblock v0.18.0)

Do you have a sample RDL that shows this working?

amykyta3, 2023-09-09:
Works fine for me with an example like this:
```
addrmap top {
    reg ro_reg {
        field {
            sw=r; hw=w;
        } ro_field;
    };
    internal ro_reg int_ro_reg;
    external ro_reg ext_ro_reg;

    reg wo_reg {
        field {
            sw=w; hw=r;
        } wo_field;
    };
    internal wo_reg int_wo_reg;
    external wo_reg ext_wo_reg;
};
```
Make sure the tool is truly picking up the correct instance of peakrdl-regblock via:
```
peakrdl --plugins
```
Sometimes multiple installations across a virtual environment or home directory install can take surprising precedence.