SystemRDL/PeakRDL-regblock #59: Address is scaled for long register (regwidth > accesswidth)

always-debug, 2023-08-07:
Hello,

When I define regwidth = 128, and accesswidth = 32 for an register array, i.e REGL[16]. The addresses of REGL is incremented by 16 bytes (regwidth / accesswidth = 4* 128 / 32) for each register. For instance, REGL[0] = 0x0000, REGL[1] = 0x0010, REGL[2] = 0x0020, ... 

I have used buffer_writes and buffer_read and wbuffer_trigger and rbuffer_trigger to handle the data mismatch between register size of data bus size so I suppose the address must be increased by the bus size. Meaning that, REGL[1] = 0x0004, REGL[2]=0x0008 but in order to read/write a complete 128b register, we need to write 4 times and reads 4 time from the same address rather than read/write to 4 different address.

Can PeakRDL support this?

Thanks

always-debug, 2023-08-07:
Hi @amykyta3 

The reason to use long registers are sometime to save memory address map. Therefore, I am expecting 
0x0000: REGL[0], bits [127:0]
0x0004: REGL[1], bits [127:0]
0x0008: REGL[1], bits [127:0]

So if I use buffer_writes and wbuffer_trigger, to write REGL[0], I will need to write 4 times (via 32b APB data) into the same address 0x0000 then trigger the write buffer to update REGL[0]. This is what I am thinking. If we need to split  REGL[0] into 4x32b ranging from 0x0000 to 0x000C, it is similar to define 4x32b separate registers and it does not help to reduce the address map (some IP has a limit address mapping space for CSR)

Thanks

philipaxer, 2023-08-09:
Hi, so far I am not actively using PeakRDL, but only silently listenting to the isssues and code progress (gread work btw!).

From an (automotive) system's architect point of view the semantics as specified in the previous post are really discouraged as they contain implicit state.  In order to support the scheme, the register file RTL needs to implement a modulus counter that keeps track how often a register has been accessed (i.e. which subpart the reg is currently read/written). 

First issue, intra-register read/write interleaving is unclear. Is the same counter used, or should the FSM be seperated?

Second issue, inter-register access interleaving is unclear. What happens if only 2 subwords are read from REGL[0] and then you continue with REG[1]. Will it start with the LSBs or continue where it left if with REGL[0]?

Third, issue is that it is very easy to accidentally get out of sync. If the RTL resets, or the firmware is restarted, the software asumption is not in-line with the HW state.

I would strongly discourage using this scheme.  If you really must limit the number of addresses, please consider a proper indirection register. 

In the example as given above, i would prefer the approach below, but i am not sure if SystemRDL/PeakRDL supports this

0x0000: N
0x0004: REGL[N], bits [31:0]
0x0008: REGL[N], bits [63:32]
0x000C: REGL[N], bits [95:64]
0x0010: REGL[N], bits [127:96]

regards
Philip

amykyta3, 2023-08-10:
Thanks @philipaxer for your detailed answer! I generally agree with you on this - such a design to stack storage elements has some intrinsic hazards that need to be considered. That said, I do recognize the challenge of having limited availability of address space, and having to resort to some more unorthodox solutions. If I were writing a register file by-hand, I can think of a few ways this could be accomplished that avoid these functional hazards.

@always-debug - The structure you describe is not supported in SystemRDL. The language has a pretty firm philosophy that every field shall be accessible by a unique address offset & distinct bit range. It is not possible to describe the bit-stacking behavior you desire natively in the language.
One option that you do have is to describe these registers as external. PeakRDL-regblock does support this, but it effectively breaks out interfaces to these individual registers and leaves it up to you to implement the special behavior.
For more details on how this mechanism works, see: https://peakrdl-regblock.readthedocs.io/en/latest/rdl_features/external.html

amykyta3, 2023-08-25:
Closing since I think this question is answered. Feel free to reopen if there is anything else to discuss.

closed 2023-08-25 by amykyta3