SystemRDL/PeakRDL-regblock #60: Lower bits of cpu addr bus not ignored with wider data bus

roowatt, 2023-08-08:
I am working on the C Headers for register access interfacing a regblock generated to Xilinx Microblaze via AXI-Lite. I will post some screenshots at the end from Vivado that shows the issue that I will try to describe...

I am using an accesswidth and regwidth of 32, I have defeined a number of registers and I am poking at them from the CPU side. I have noticed the following. It looks like only writes that are regwidth aligned are successful, consider the following register that is located at offset 0x20...

```
	reg {
		field {
			desc="Set bit high to enable the processor load interface.
			Preamble generation can not take happen when load interface is enabled.";
		} LOAD[24:24] = 0;
		field {
			desc="Set bit high to enable hybrid preamble generation.";
		} HYBRID_MODE[16:16] = 0;
		field {
			desc="Set bit high to enable postfix generation, only valid in HYBRID_MODE";
		} HYBRID_POSTFIX[8:8] = 0;

		field {
			singlepulse;
			desc="Writing a 'one' to this register will start preamble generation";
		} START[0:0] = 0;
	} control0;
```
As can be seen, there are four 1 bit fields that are the LSB of the 4 bytes of the 32 bit register.

If I use a uint8_t pointer to access these fields, the writes to the first field succeeds but fails to the remaining three fields. Despite the WSTRB having the expected values. However, if I mask off the lower 2 bits of the CPU address then things work as I expect.

It seems to me that the lower 2 bits of the address bus should be ignored when generating the `decoded_reg_strb.xx.yy` signal.

Am I barking up the wrong tree here?

** Lower 2bits not masked**
[image]

** Lower 2bits masked**
[image]

I can provide additional information or clarification as needed.

roowatt, 2023-08-08:
Version info is
```
peakrdl                0.8.0
peakrdl-html           2.10.1
peakrdl-ipxact         3.4.1
peakrdl-regblock       0.16.0
peakrdl-systemrdl      0.3.0
peakrdl-uvm            2.3.0
```

amykyta3, 2023-08-10:
Yep definitely a bug. CPUIF implementations for AXI4-Lite, and probably also APB should ignore the lower bits of the address bus in order to properly handle unaligned accesses.

roowatt, 2023-08-10:
"Unaligned Access"... that is a far simpler way of describing the issue. :)

I guess I was so focussed on packing registers so they could be efficiently accessed multiple ways that I didn't consider it to be unaligned as such.

amykyta3, 2023-08-25:
Fixed in v0.18.0

closed 2023-08-25 by amykyta3