SystemRDL/PeakRDL-regblock #63: Vivado Synthesis Compatability

roowatt, 2023-09-07:
Using latest peakrdl-regblock v0.18.0

I have an issue with synthesis compatability with Vivado 2023.1 when using asynchronous resets (I haven't checked other kinds).

If I have afield without a reset value (ie No reset) then the Vivado synthesiser chokes with a fatal error.

A simplified rdl def is as follows...
```
addrmap test_case {
name = "Xilinx Vivado Test Case";

    default sw = rw;
    default hw = r;	

    default regwidth = 32;
    default accesswidth = 32;

    addressing = regalign;

	regfile  {
	    default sw = rw;
	    default hw = r;	

		reg {
			field {} field0[16] = 0;
			field {} field1[16] ;
		} reg0;
	} regfile0;
};
```

Logfile below...
```

*** Running vivado
    with args -log test_case.vds -m64 -product Vivado -mode batch -messageDb vivado.pb -notrace -source test_case.tcl

****** Vivado v2023.1 (64-bit)
  **** SW Build 3865809 on Sun May  7 15:04:56 MDT 2023
  **** IP Build 3864474 on Sun May  7 20:36:21 MDT 2023
  **** SharedData Build 3865790 on Sun May 07 13:33:03 MDT 2023
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
    ** Copyright 2022-2023 Advanced Micro Devices, Inc. All Rights Reserved.

source test_case.tcl -notrace
create_project: Time (s): cpu = 00:00:07 ; elapsed = 00:00:07 . Memory (MB): peak = 2067.078 ; gain = 122.992 ; free physical = 10267 ; free virtual = 33520
Command: read_checkpoint -auto_incremental -incremental /home/cameron/tmp/peakrdl_test_case/peakrdl_test_case.srcs/utils_1/imports/synth_1/test_case.dcp
INFO: [Vivado 12-5825] Read reference checkpoint from /home/cameron/tmp/peakrdl_test_case/peakrdl_test_case.srcs/utils_1/imports/synth_1/test_case.dcp for incremental synthesis
INFO: [Vivado 12-7989] Please ensure there are no constraint changes
Command: synth_design -top test_case -part xcvu095-ffva2104-2-e
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xcvu095'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xcvu095'
INFO: [Designutils 20-5440] No compile time benefit to using incremental synthesis; A full resynthesis will be run
INFO: [Designutils 20-4379] Flow is switching to default flow due to incremental criteria not met. If you would like to alter this behaviour and have the flow terminate instead, please set the following parameter config_implementation {autoIncr.Synth.RejectBehavior Terminate}
INFO: [Synth 8-7079] Multithreading enabled for synth_design using a maximum of 4 processes.
INFO: [Synth 8-7078] Launching helper process for spawning children vivado processes
INFO: [Synth 8-7075] Helper process launched with PID 698416
---------------------------------------------------------------------------------
Starting Synthesize : Time (s): cpu = 00:00:04 ; elapsed = 00:00:04 . Memory (MB): peak = 2492.703 ; gain = 383.738 ; free physical = 9610 ; free virtual = 32863
---------------------------------------------------------------------------------
INFO: [Synth 8-6157] synthesizing module 'test_case' [/home/cameron/projects/kakadu/kakadu-asic/peakrdl/build/test_case.sv:4]
INFO: [Synth 8-6157] synthesizing module 'axi4lite_intf' [/home/cameron/projects/kakadu/kakadu-asic/peakrdl/lib/axi4lite_intf.sv:1]
INFO: [Synth 8-6155] done synthesizing module 'axi4lite_intf' (0#1) [/home/cameron/projects/kakadu/kakadu-asic/peakrdl/lib/axi4lite_intf.sv:1]
INFO: [Synth 8-6157] synthesizing module 'axi4lite_intf' [/home/cameron/projects/kakadu/kakadu-asic/peakrdl/lib/axi4lite_intf.sv:1]
INFO: [Synth 8-6155] done synthesizing module 'axi4lite_intf' (0#1) [/home/cameron/projects/kakadu/kakadu-asic/peakrdl/lib/axi4lite_intf.sv:1]
ERROR: [Synth 8-91] ambiguous clock in event control [/home/cameron/projects/kakadu/kakadu-asic/peakrdl/build/test_case.sv:294]
ERROR: [Synth 8-6156] failed synthesizing module 'test_case' [/home/cameron/projects/kakadu/kakadu-asic/peakrdl/build/test_case.sv:4]
---------------------------------------------------------------------------------
Finished Synthesize : Time (s): cpu = 00:00:05 ; elapsed = 00:00:05 . Memory (MB): peak = 2558.672 ; gain = 449.707 ; free physical = 9524 ; free virtual = 32777
---------------------------------------------------------------------------------
synthesize failed
INFO: [Common 17-83] Releasing license: Synthesis
14 Infos, 0 Warnings, 0 Critical Warnings and 3 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Vivado Synthesis failed
INFO: [Common 17-206] Exiting Vivado at Thu Sep  7 15:54:44 2023...
```

roowatt, 2023-09-08:
Removed extraneous log file details.

amykyta3, 2023-09-08:
Thanks!
I have confirmed the issue and the fix will be in the next release.

amykyta3, 2023-10-13:
Published in 0.19.0

closed 2023-10-13 by amykyta3