SystemRDL/PeakRDL-regblock #64: Support for packed structs

leolitenstorrent, 2023-09-14:
Hello Alex,

May PeakRDL have an option (for example, in the exporter) to support packed structs? It would be useful, for example, to assign the entire register as a whole in a testbench, etc.

My team would appreciate it very much!

Thank you!

amykyta3, 2023-09-14:
Nothing prevents you from assigning unpacked structs as a whole as well.

Try using the default assignment mechanism for initializing:
```systemverilog
packed_struct = '{default: '0};
```

Or use the streaming operator if you want to assign a vector into a packed struct:
```systemverilog
packed_struct = {>>{64'h12345678ABCD9876}};
```

I expect these widely supported methods will work for you.

leolitenstorrent, 2023-09-14:
Noted. Thank you!

closed 2023-09-14 by leolitenstorrent