SystemRDL/PeakRDL-regblock #65: Proposal to use `logic` instead of `wire` for module inputs

jeras, 2023-09-14:
I do not know if there is any strong recommendation for using `logic` for all RTL signals in SystemVerilog, but I and many open source projects do so. For example the [Pulp project](https://github.com/pulp-platform/snitch/blob/master/hw/ip/snitch/src/snitch.sv).

One advantage of `logic` over `wire` is checking for multiple drivers, but I am not sure if this is relevant for inputs.

I personally use wire only for tristate signals like GPIO, I2C, 1-Wire, ... in top level modules (chip IO pins) and test benches.

I can prepare a pull request if you wish to make this change.

jeras, 2023-09-20:
I forgot to mention, the required Python package `venv` is not mentioned in the instructions.

amykyta3, 2023-09-29:
In SystemVerilog, inputs are implicitly `wire` net types and implicitly `logic` data types. Therefore for inputs, changing the port declaration from `input wire` to `input logic` would have no effect since both are implicitly `input wire logic`.
Furthermore, some users will use the directive ``default_nettype none` to disable implicit wires. If inputs are declared as `input logic`, then they will not inherit a net type of wire anymore which would result in a compile error.

See section 23.2.2.3 of the IEEE Std 1800-2017 SystemVerilog standard:
[image]

amykyta3, 2023-09-29:
Not a problem. I think in this case, some situations may cause compile errors for some users, so I would prefer to not make this change.

closed 2023-09-29 by amykyta3