SystemRDL/PeakRDL-regblock #69: using write/read suffixes instead of input/output for HW interface

jeras, 2023-10-07:
The current implementation uses input/output based suffixes to distinguish the two HW port structures:
```systemverilog
        input Example_pkg::Example__in_t hwif_in,
        output Example_pkg::Example__out_t hwif_out
```
Instead I would recommend using `write`/`read` (or a shorthand like `w`/`r`) based suffixes:
```systemverilog
    input   Example_pkg::Example__write_t hwif_write_i,
    output Example_pkg::Example__read_t hwif_read_o
```
And the instance:
```systemverilog
    .hwif_write_i (hwif_w),
    .hwif_read_o  (hwif_r)
```

The first reason is consistency with the the SystemRDL standard, which describes read/write access to fields from the HW side.

The second reason is to have more consistent and intuitive signals across the RTL hierarchy.
I encountered this problem with the Wishbone system bus protocol specification, where write read data ports were named `data_o` and `data_i`.
This convention creates issues when connecting master and slave modules instantiated in the same hierarchical level, since write data is an output on the master, but an input on the slave (and vice versa for the read data). So what would be the right suffix to use for the signals connecting the master to the slave.
Using write/read suffixes solves this issue, and it can also reduce the renaming of signals on port connections, thus it can make it easier to follow signals across hierarchies using text editor search.

amykyta3, 2023-10-07:
Renaming the regblock ports to be suffixed by "read" and "write" would be inappropriate. Often there are output signals that are associated with read-only fields (such as swacc), input signals that are associated with writable fields (write-clear interrupts, hwset/hwclear), or input signals that are not associated at all with any field such as user signals. Using suffixes that associate the ports with read/write functions would be misleading or even completely inaccurate.

Regarding in/out suffxixes, the current hwif ports already end in `*_in` and `*_out`, so changing or adding additional "i" or "o" suffixeds is not needed. There are many conflicting coding styles out there, so it is impossible to satisfy all of them

jeras, 2023-10-08:
Sorry, I wrote this late at night. As soon as I laid into bed I remembered you combined data and control signals into the same structure.
GitHub should have a way to create issue drafts, so I can review them after sleeping on it. GitLab at least remembers the draft issue contents in a cookie if you close the browser.

closed 2023-10-08 by jeras

amykyta3, 2023-10-08:
Hehe no problem!
All great discussion nonetheless