SystemRDL/PeakRDL-regblock #72: Atomic accesses to memory components

apstrike, 2023-11-06:
Is there a schema to create SW atomic accesses to memory components where the `memwidth` is greater than the `cpuif bus width`? 
I notice that the external memory bus always has the same width as the `cpuif` rather than `memwidth` and I think this may be somewhat counter-intuitive. I've attempted to use the `buffer_reads` and `buffer_writes` properties on virtual registers withing the `mem` component but saw no difference in the result. If this hasn't been though of yet (and I think this atomic memory read/writes would be quite a useful benefit to the peakrdl project) may I suggest the following approach:

- change buses to the external memory to be `memwidth` bits wide. 
- For write requests: if a user doesn't wish to have atomic accesses to the memory element, then the write data gets shifted based on the offset within the memory's word on the `wr_data` signal and the same goes for the bitenables on the `wr_biten` bus. If a user does wish to have atomic accesses to the memory element, then the `buffer_writes` property can be set on the memory component and a similar buffer storage schema that is used for registers can be applied. The one caveat is that the write requests to the memory must be for the same word in the memory until the write is complete and an atomic write can be issued. if a write is recieved for a different word in the memory, then the initial atomic write is abandoned. If the chosen `cpuif` bus type has a write error signal, then that could be set to indicate an issue has occurred.
- For read requests: if a user doesn't wish to have atomic accesses to the memory element, then on a read request, we store the offset within the word that we expect the read data to come from. When a `rd_ack` is received we use said cache of the word offset to shift the `rd_data` to align with the read address requested on the cpuif bus. If a user does wish to have atomic accesses to the memory element, then the `buffer_reads` property can be applied on the memory component and a similar buffer storage schema that is used for registers can be applied. Here I'd expect, similar to the register model, to cache the memory word upon a read of the lowest bits of the memory word and use that cache for reads until a new word is requested to be read.

I know that this approach has some dangers. For example, if a user attempts to poll a field/bit with a memory word this value would be a cached value when using the `buffer_reads` property. But I think it may be on the user to ensure that the correct properties/access types are set when designing the memory interface for their purpose.

amykyta3, 2023-11-07:
This is a little bit out of scope for the regblock generator.

When you instantiate a `mem` component, the regblock generator doesn't actually make any inference about what type of component it is (an `addrmap` or external `regfile` will result in the same thing). The only thing that matters in this situation is that it is an external component as defined by SystemRDL which means that the implementation of the component is outside of the register block currently being generated. As soon as the regblock generator sees a component marked as `external`, it will actually skip inspecting into its contents entirely.

You make a good observation regarding the external port's width - it will always match the accesswidth of the CPU interface. This is intentional since in my opinion, an external access port should not make any assumptions about the implementation of the component it is connecting to. Instead, I felt it was best to preserve the internal datapath as much as possible, and let the designer attach an adapter that suits the specific application. This could be an adapter that builds atomic accesses (which requires knowledge of the CPU interface access width), or one that performs partial ones via write strobes.

Hope that explanation helps.

apstrike, 2023-11-07:
Hmm yes ok I see. Thanks for the explanation.

closed 2023-11-07 by apstrike