SystemRDL/PeakRDL-regblock #75: Intel Quartus support

eruanno123, 2023-11-24:
This is more of a question rather than a bug report. As of 0.19.0, does the `regblock` plugin work with Intel Quartus synthesis tool?

From my observation, there is an issue with the implicit `generate` blocks. For example, this syntax (apparently being a part of SystemVerilog-2005) does not work:

```systemverilog
for (genvar i = 0; ...) begin
    <...>
end
```

What does work instead is the 'old-school' Verilog approach:

```systemverilog
genvar i;
generate for (i = 0; ...) begin : g_<label>
    <...>
end
endgenerate
```

I am interested in adding some sort of "legacy" support, however, I might need some hints. For now, I will try to figure out for my internal needs, but later I might share this as a pull request. First, I would like to exclude early things like:

1. Does the 'regblock' plugin work with Intel Quartus and I am just stupid and cannot configure Quartus properly?
2. Does Intel Quartus have some command line option, that enables newer SystemVerilog standards? 

Regarding the source code, what I see is a class called `LoopBody`. My first idea is to implement a separate class `GenerateLoopBody`. which would cover the special needs of the old-fashioned syntax and provide it wherever needed. By default Quartus seems to be unhappy about missing generate block labels, so there needs to be some loop naming algorithm. Also, the standalone `genvar` declaration must appear once and cannot collide with other variables in the module. Perhaps, the entire change could be controlled with an additional command line flag, like `--language-tweaks=quartus`.

Does it all make any sense? Any hints regarding how it could be implemented?

Thanks.

amykyta3, 2023-11-26:
`generate`/`endgenerate` blocks became officially optional starting in SystemVerilog 2009 (Unclear in SV2005. It doesn't explicitly say whether they are required or optional)

Which version of Intel Quartus are you using?
Quartus should have no issue with this syntax unless you are using a *very* old version: https://www.intel.com/content/www/us/en/docs/programmable/683236/23-3/verilog-and-systemverilog-synthesis-support.html

Is it possible you are marking the generated output as Verilog and not SystemVerilog? Some tools will make the distinction where "Verilog" implies Verilog 2001 and older and SystemVerilog implies 2005 or newer.

closed 2024-01-06 by amykyta3

amykyta3, 2026-01-29:
Please open a new issue rather than resurrecting old ones. it makes it very difficult to keep track of requests and changes otherwise.
Can you also describe what the compatibility issue is? I can somewhat guess from the patch contents, but it is not clear to me what the original intent is.

KeatonClark, 2026-01-29:
Sorry, I didn't mean to resurrect this issue, and it is not really an issue as it is an incompatibly with an old standard that could be a pain to fully support.
As the original issue stated, the way this tool creates genvar loops:

```
for (genvar i = 0; ...) begin
    <...>
end
```

is incompatible with the standard that Quartus Prime Lite 25.1 uses for System Verilog (2005).
The patch above generates the loops as:

```
generate begin
genvar i0;
for (i0 = 0; ...) begin : g_<random_label>
    genvar i1;
    for (i1 = 0;...) begin : g_<random_label>
    end
end
end endgenerate
```

I can open a new issue and a pr for this if it is something you are interested in adding and supporting, but because it is so outdated I was thinking it could just live here for when someone looks for "quartus" in the closed issues.