SystemRDL/PeakRDL-regblock #76: Raising an error for unauthorized read or write

imerkado91, 2023-12-11:
Is it possible to add an indication for unauthorized read/write (or maybe it already exists and I missed it...)

For example, let's say that a field/reg is defined as sw=w, but the SW is trying to read this field/reg, we would like to see an indication/error assertion for it (and the other way around with sw=r when it tries to write).
is it supported? Is there a way to add this?

amykyta3, 2024-01-12:
Currently not supported, but it is something I was considering adding eventually.

There would be some limitations with this - like only issuing an error if a write occurred on a register that contains only read-only fields, and vice-versa. SystemRDL lets you define registers that have mixed-type fields so it would not be practical to assert this error on a per-field basis.

NajamKhalil, 2024-01-22:
Will you kindly share the list of condition on which this error may get asserted so that in case some one try to implement this at there end it would be easy for them.If i am not the AMBA APB specs says this error assertion is not necessary as per specs,"Completers are not required to support PSLVERR. Where a Completer does not include PSLVERR, the appropriate input to the Requester is tied LOW."And you are doing the same thing by assigning it to zero.

amykyta3, 2024-01-22:
Yes currently it is tied to a constant zero.

NajamKhalil, 2024-01-22:
Any idea about the list of conditions on which this error may get asserted.?
Since i am unable to find it on the specs of AMBA apb protocol handbook.

amykyta3, 2024-01-22:
AMBA does not specify this. It is up to the implementation of a register file device to decide.
A PSLVERR may be due to accessing an address without an assigned register, or writing a register that is read-only, or vice-versa. It really depends on what requirements a particular project needs and what is appropriate.
For PeakRDL, if implemented, I would make it configurable.

NajamKhalil, 2024-01-22:
Got it.
Thanks for your support.

amykyta3, 2025-10-26:
Implemented via #168