SystemRDL/PeakRDL-regblock #83: Issue: Error in RegblockExporter.export regarding buffer_writes

xcfuisnewhere, 2024-01-17:
Hi everyone, can you please help me about this problem below:
1. When I use *RegblockExporter.export()* function to generate rtl code, I got an error like this: 
```log
Traceback (most recent call last):
  File "C:\W_Work\2024_01_04_RegTool2.0\Tool\2.DATA\ipxact2rtl.py", line 42, in <module>
    rtl.export(
  File "C:\Users\xcfu\AppData\Local\Programs\Python\Python310\lib\site-packages\peakrdl_regblock\exporter.py", line 128, in export
    self.ds = DesignState(top_node, kwargs)
  File "C:\Users\xcfu\AppData\Local\Programs\Python\Python310\lib\site-packages\peakrdl_regblock\exporter.py", line 252, in __init__
    DesignScanner(self).do_scan()
  File "C:\Users\xcfu\AppData\Local\Programs\Python\Python310\lib\site-packages\peakrdl_regblock\scan_design.py", line 58, in do_scan
    RDLWalker().walk(self.top_node, self)
  File "C:\Users\xcfu\AppData\Local\Programs\Python\Python310\lib\site-packages\systemrdl\walker.py", line 159, in walk
    self.walk(child, *listeners)
  File "C:\Users\xcfu\AppData\Local\Programs\Python\Python310\lib\site-packages\systemrdl\walker.py", line 150, in walk
    self.current_action = self.do_enter(node, listener)
  File "C:\Users\xcfu\AppData\Local\Programs\Python\Python310\lib\site-packages\systemrdl\walker.py", line 193, in do_enter
    new_action = listener.enter_Reg(node) or WalkerAction.Continue
  File "C:\Users\xcfu\AppData\Local\Programs\Python\Python310\lib\site-packages\peakrdl_regblock\scan_design.py", line 97, in enter_Reg
    self.ds.has_buffered_write_regs = self.ds.has_buffered_write_regs or bool(node.get_property('buffer_writes'))
  File "C:\Users\xcfu\AppData\Local\Programs\Python\Python310\lib\site-packages\systemrdl\node.py", line 397, in get_property
    raise LookupError("Unknown property '%s'" % prop_name)
LookupError: Unknown property 'buffer_writes'
```

2. My python code is as follows:
```python
rdlc = RDLCompiler()
ipxact = IPXACTImporter(rdlc)

try:
    ipxact.import_file("xxxx.xml")
    root = rdlc.elaborate()
    regmap = root.find_by_path('cmsdk_apb_watchdog__APB_Watchdog_Slave_MM.addressBlock0')
except RDLCompileError:
    sys.exit(1)

rtl= RegblockExporter()
rtl.export(
    regmap, "./rtl/",
    cpuif_cls=AXI4Lite_Cpuif
)
```

amykyta3, 2024-01-17:
You are missing the step where UDPs required for the regblock generator get registered. See the example here: https://peakrdl-regblock.readthedocs.io/en/latest/api.html#example

Also, be sure to check out the [PeakRDL command line tool](https://peakrdl.readthedocs.io/en/latest/gallery.html) - it ties together all of the PeakRDL tools into one easy to use toolchain.

xcfuisnewhere, 2024-01-18:
Thank you so much for your help! I have another question to ask. When editing IP-XACT and SystemRDL files with a text editor, it doesn't feel very intuitive. I prefer to edit each entry in a more tabular format. Do you have any recommended editors or tools for this?

amykyta3, 2024-01-18:
If you use VSCode, I have made a SystemRDL extension that adds syntax highlighting and templates. Its still a text editor, but at least it can be easier to read. Once you get used to the language, it becomes pretty intuitive. I have a tutorial here if you need some guidance: https://peakrdl.readthedocs.io/en/latest/systemrdl-tutorial.html

I dont have a good recommendation for a tabular method. Partly because I prefer just editing RDL directly.

Agree that editing IP-XACT files is an absolute pain. I avoid it whenever possible. I know of the Kactus2 tool which is a graphical editor for IP-XACT, but I have not used it too much. Usually I'll generate IP-XACT from a SystemRDL file if I absolutely need to. (via `peakrdl ipxact`)

xcfuisnewhere, 2024-01-18:
Thank you very much！ Your suggestions have been immensely helpful to me.

closed 2024-01-18 by xcfuisnewhere