SystemRDL/PeakRDL-regblock #85: Support for iverilog

shareefj, 2024-01-21:
Hey there,

I'm looking for a SystemRDL Verilog exporter that outputs Verilog or SV that is readable by Icarus Verilog.  The Regblock exporter seems to output RTL with a number of styles that iverilog can't cope with, including:

- the use of unpacked structs
- using automatic scope variables in always blocks
- using struct members in always blocks

That's as far as I got in my analysis.  I just wanted to check if you had any plans on adding support for iverilog before I looked further.  I think most of the verification style syntax issues would go away by using sub-modules for the field logic, similar to the OpenTitan regtool.  The struct declarations are easy to fix.

Any plans for iverilog support on your horizon?

amykyta3, 2024-01-21:
Supporting iverilog in its current state would be difficult to do and likely not a priority at this time. Notably - use of struct members in always blocks is perfectly LRM compliant and is supported by the majority of other tools. Re-architecting the regblock generator to avoid this would be a huge undertaking.
Currently, iverilog doesn't really have very good support for SystemVerilog yet. I would recommend opening a support ticket with the iverilog maintainers to see if they can extend support for these features. Looking at their issue list, it looks like there is a lot of recent activity regarding SV features, so perhaps some of these things will be addressed soon: https://github.com/steveicarus/iverilog/issues

In the meantime, verilator should work for the generated RTL. I do have an open ticket (#41) tracking some other limitations in verilator, but all of them are regarding constructs in my testbench code and not the generated RTL.
Another option is to use the free version of the Questa simulator: https://github.com/SystemRDL/PeakRDL-regblock/tree/main/tests#questa

solomatnikov, 2024-01-22:
If the tool doesn't support language constructs introduced ~20 years ago, does it make sense to continue to use such a tool?

amykyta3, 2024-01-22:
I'm not familiar with iverilog too much, but I always thought of it as a Verilog simulator and not SystemVerilog. My guess is that the addition of SystemVerilog support is more of a recent development so maybe they haven't gotten around to some of the slightly less popular features.

It's still worth submitting a bug report for them - at the very least, it signals to them which features are useful and need attention, even if they may not get around to it soon. As the maintainer of the PeakRDL projects, I hugely appreciate each bug report or feature request since it gives me valuable feedback on what users find important (even though I have some feature requests I haven't touched in years!)

shareefj, 2024-01-22:
@amykyta3 Yep, the SV support is new and getting better but still lags commercial tooling.

@solomatnikov A lot of opensource projects have no option but to use Verilator or iverilog.  And just because it's in the LRM, doesn't always mean that vendors support it.  Go back 10 years and the support for SV by even the major EDA vendors was still flakey.  A lot of the syntax used in this tool seems to be more of a verification SV style which will always cause problems.

I'll get round to opening a ticket on iverilog at some point but it looks like I'll try modifying the exporter for now.

Cheers.

closed 2024-01-22 by shareefj