SystemRDL/PeakRDL-regblock #90: infer accesswidth when only external components in RDL

saberxt, 2024-03-21:
Does this exporter have some configuration related to this topic?

Here is the situation, we have 3 rdl files, each of them is an addrmap contains some regs which accesswidth is 16bit. And we have another rdl file, which is an addrmap with the 3 addrmaps before instantiated in it. So the last rdl file only contains external components. When we generate RTL from this, we will get a warning shows peakrdl is assuming the cpuif bus width is 32bit. This is obviously not compatible with those 3 external rdls.

Do you have any idea to solve this?

amykyta3, 2024-03-30:
Is your intent to generate a single regblock that implements the three addrmaps internally?
If so, you could change the three child `addrmap` components to `regfile` components so that they are treated as internal.

Or is the intent to actually have three external addrmaps and no internal contents in the regblock?

saberxt, 2024-03-30:
Yes, we are going to have three external addrmap, because we need three rtl files in different places, and they share the same address decoder which is the top one. Now we just put one hack register in the top one to tell the top module that we need 16 bit width.

amykyta3, 2024-03-31:
Ah I see. You're effectively using it as a bus decoder.

A dummy register is probably a good workaround for now...

I think the long-term solution for this will be that I build a proper decoder/crossbar generator that is separate from the regblock generator. The regblock generator isn't really optimized for this type of use-case, however your use of the generator as an address decoder is certainly clever!
Building a crossbar generator is certainly something I had considered doing in the past already.

saberxt, 2024-03-31:
Thank you for your response. 

I know that use regblock as decoder is not the intention of it, but sometimes it could work, if our access width is 32bit, then we don't need any dummy register.

As you already have the idea to build such a new module, we are looking forward to it coming true.

closed 2024-11-29 by amykyta3