SystemRDL/PeakRDL-regblock #91: Request to change from typdef enum int to typedef enum logic to resolve Lint Issues.

amullick007, 2024-03-26:
Hi,

We are currently getting some Lint failures due to the usage of "typedef enum int" statements in the generated RTL files from RDL for Register index.

Can we change these to "typedef enum logic" to keep Lint happy?

Regards,
Avirup

amykyta3, 2024-03-27:
Can you share an example?
I don't know of any uses of `typedef enum int` anywhere. Are you referring to the enum definitions that are `typedef enum {...} name_t`?

calebofearth, 2024-03-30:
@amykyta3, I think the original occurrence of `typedef enum int` may have manually been inserted into a reg file rather than generated. But we are seeing the `typedef enum {...}` syntax you mentioned - [here's](https://github.com/chipsalliance/caliptra-rtl/blob/user/dev/michnorris/lint_fix/src/ecc/rtl/ecc_reg_pkg.sv#L446:L450) an example generated from peakrdl-regblock version 0.21.0. This defaults to `typedef enum int` per IEEE 1800-2017, 6.19, so it's effectively the same issue.
For synthesis, these should be 4-state variables.
Thanks.

amykyta3, 2024-03-30:
Yep aware of that. Just checking in case something else was going on in a different part of the tool.
Will change so that it declares the enum based on the bitwidth of the largest enum member. Note that with RDL enums, it is not possible to size it based on the field it is assigned to since RDL enums can potentially be shared across multiple different sized fields.

amykyta3, 2024-03-30:
Will be fixed in upcoming release

closed 2024-03-30 by amykyta3

calebofearth, 2024-03-30:
Excellent, thanks for the quick fix!