SystemRDL/PeakRDL-regblock #92: Casting For-Loop Iterator to Match Address Width in Decoding Logic

leolitenstorrent, 2024-04-02:
Dear PeakRDL Development Team,

```
    //--------------------------------------------------------------------------
    // Address Decode
    //--------------------------------------------------------------------------
    typedef struct {
        // Code not shown
    } decoded_reg_strb_t;
    // Code not shown

    always_comb begin
        // Code not shown
        for(int i0=0; i0<32; i0++) begin
            decoded_reg_strb.REGISTER_ARRAY[i0] = cpuif_req_masked & (cpuif_addr == 11'hf8 + i0*11'h8);
        end
    end
```

In the SV code generated above for address decoding, the variable `i0` has bitwidth 32 whereas the address has bit width 11 (in this case). This results in the Lint warning of "Unequal length in comparison operator". May the team consider resolving this? For example, `i0*11'h8` can be changed to `11'(i0)*11'h8`.

Thank you!

amykyta3, 2024-12-20:
Easy enough to add.
Width cast will be included in the next upcoming release.

amykyta3, 2024-12-20:
Implemented in v0.23.0

closed 2024-12-20 by amykyta3