SystemRDL/PeakRDL-regblock #95: Files generated by this project do not work with Verilator

paul-demo, 2024-04-05:
Verilator doesn't allow unpacked structs in the top level, so register files generated by PeakRDL-regblock cannot be simulated.

Even changing them to packed structs does not work because inside of those structs, you have unpacked structs of other structs. For example: if you have an array of registers in your register file, PeakRDL-regblock generates an unpacked array inside the struct; so even if we change the structs to packed, it results in another error "cannot have unpacked element inside packed element".

paul-demo, 2024-04-06:
It could be that it just can't be at the top level of a verilator build, which actually isn't a big limitation. Once I get a wrapper module assembled, I'll try it out with the unpacked structs.

amykyta3, 2024-04-06:
I haven't done any unit-testing with Verilator yet, but from what I understand it should work. There are a few limitations with Verilator that prevent me from also using in unit-tests due to testbench constructs that I use (#41), but it looks like the developers are pretty close to expanding support needed for that too.

paul-demo, 2024-04-06:
Hooray! It seems to work fine leaving them as unpacked structs as long as you are not using the generated file as the top level (which I was doing originally, just to simulate it to see what artifacts this project produces).

closed 2024-04-06 by paul-demo