SystemRDL/PeakRDL-regblock #96: Timing of swmod seems 1 cycle too early

paul-demo, 2024-04-08:
I have a volatile register as explained here: https://github.com/SystemRDL/PeakRDL-regblock/issues/94

```
    reg {
        field {
            sw = w;
            hw = r;
            swmod;
        } read_issue[31:0];

        field {
            sw = r;
            hw = w;
        } read_observe[31:0];
    } read[2];
```

I was expecting `swmod` to come out at the same time as `read_issue`, but it seems to come out one cycle before the data value:
[image]

Is this the expected behavior for swmod? Is the timing of these various optional signals documented by Accellera or is it a design choice of peakrdl-regblock how they should work?

amykyta3, 2024-04-09:
Accellera does not specify the behavior in their spec. This is intentional design decision for this generator. See #19 for an explanation.

paul-demo, 2024-04-09:
Thanks; that's fine just something I noticed compared to other generators which generally have a "strobe" output like this in the same cycle as the new data value.

closed 2024-04-09 by paul-demo