SystemRDL/PeakRDL-regblock #97: External mem state machine gets confused by rd_ack that may result from READ_FIRST BRAMs

paul-demo, 2024-04-08:
If I have a BRAM configured in NO_CHANGE mode, then hooking that up as a memory backing an `external mem` declaration works fine in the generated peakrdl-regblock logic. The read acknowledge looks something like `rd_ack <= bram_en & ~|bram_we;` and the write acknowledge is `wr_ack <= bram_en & |bram_we;`. Just like Xilinx BRAM18/36, the RAM implementation itself needs the enable to be asserted for either reads or writes to occur, and then writes additionally need to assert the `we` bit(s).

However, if I have a READ_FIRST ram, then every write is also a read. If I simply register the read-enable through the appropriate number of cycles to produce `rd_ack` for peakrdl, and I pipeline the write enable through to produce `wr_ack`, then this confuses the state machine generated by PeakRDL and results in an `axil_rvalid` output for memory writes and there is no `axil_bvalid` response generated even though `wr_ack` pulsed high as expected.

Basically, this confuses the peakrdl-regblock logic:
`rd_ack <= bram_en;`
`wr_ack <= bram_en & bram_we;`

I saw that the generated code attempts to produce some errors to alert you that peakrdl-regblock is incompatible with this situation, but they did not fire for me. The generated code includes this block of code:

```verilog
    always_ff @(posedge clk) begin
        if(rst) begin
            external_pending <= '0;
        end else begin
            if(external_req & ~external_wr_ack & ~external_rd_ack) external_pending <= '1;
            else if(external_wr_ack | external_rd_ack) external_pending <= '0;
            assert(!external_wr_ack || (external_pending | external_req))
                else $error("An external wr_ack strobe was asserted when no external request was active");
            assert(!external_rd_ack || (external_pending | external_req))
                else $error("An external rd_ack strobe was asserted when no external request was active");
        end
    end
```

Changing my `rd_ack` implementation to follow NO_CHANGE implementation works around this issue, but it would be nice if peakrdl-regblock did not issue [read|write] responses when there are no outstanding [read|write] requests, and if the logic did not get confused by receiving a `rd_ack` in addition to receiving a `wr_ack`.

amykyta3, 2024-04-09:
I'm a little confused as to how the Xilinx BRAM implementation has any affect on this. The READ_FIRST, NO_CHANGE, and other modes of BRAM are specific to how the RAM handles collisions.
Or perhaps I am not understanding something.

Can you share a waveform that demonstrates this? Please include the relevant `hwif_in` and `hwif_out` signals, as well as the signals that interact with the RAM.

For each cycle that `hwif_out..req` is asserted, there must be exactly one corresponding clock cycle where `hwif_in..rd_ack` or `hwif_in..wr_ack` is asserted depending on whether the `req` was a read or write operation.

paul-demo, 2024-04-09:
I just had a rd_ack which pulses high every time the BRAM output changes. I sort of typed this out in paragraph form above, but basically something like this:

```verilog
always_ff @(posedge clk) begin
  if (READ_FIRST) begin
    rd_ack <= bram_en;
  end else if (NO_CHANGE) begin
    rd_ack <= bram_en & !bram_we;
  end  // else if WRITE_FIRST, etc.

  wr_ack <= bram_en & bram_we;
end
```

This definition of READ_FIRST confuses the peakrdl-regblock logic because it generates a rd_ack when the regblock has not issued a read, and this results in an erroneous read response and a non-existent write response. The motivation of having rd_ack like this is so that it always pulses high in the same cycle that a new value appears at the BRAM output.

Like the other issue I asked about, it's fine to have it this way just something a bit peculiar that we have to be aware of since the additional rd_ack will hang the bus.

paul-demo, 2024-05-08:
I thought about it some more and think it's probably reasonable that rd-ack only occurs for reads and not for writes (even if the read value is changing during the write). Closing this issue.

closed 2024-05-08 by paul-demo