SystemRDL/PeakRDL-regblock #104: Put pragmas around assertions

Blebowski, 2024-04-28:
Hi,

when trying to synthesize the generated register map with Design compiler, I get warnings due to
assertions that check the generated block behavior:

```
Line 1289: Warning:  /tmp/dummy_block.sv:83: The construct 'immediate assertion' is not supported in synthesis; it is ignored. (VER-708)
```

I think it would be good to put `synthesis translate on/off` around these assertions.

amykyta3, 2024-05-01:
Comment pragmas are not ideal since they have extremely poor standardization and have numerous variations across different toolchains, so they are generally not considered good practice.

I would be open to wrapping these assertions with `ifndef SYNTHESIS` preprocessor directives instead.

Blebowski, 2024-05-01:
Agreed, this is a better approach. `SYNTHESIS` define sounds good.

amykyta3, 2024-12-20:
Implemented in v0.23.0