SystemRDL/PeakRDL-regblock #105: Parity on R/W registers

Blebowski, 2024-04-28:
Hi,

I think an usefull feature would be adding Parity protection "per-word" on registers that have storage elements inside
the generated RTL code. When any of the interfaces would be writing such register, the stored Parity would
get updated.

For error detection there could be following outputs:
- Serial output indicating an error ocurred in any of the registers (OR of error checks on all registers). This is to be able to reflect on the error directly by HW (e.g. some sort of alarm monitor).
- When CPU is reading a register, return "slv_err" on the APB bus.

amykyta3, 2024-05-01:
This feature already exists. See the SystemRDL `paritycheck` property.
Regarding slverr, this request is captured by an existing issue #76

Blebowski, 2024-05-01:
Thanks, this is awesome. I did not know about this feature of System RDL. Its nice it works also for single bit fields where it gives you complementary encoded bit.

closed 2024-05-01 by Blebowski