SystemRDL/PeakRDL-regblock #106: Fine-grained access rights

Blebowski, 2024-04-28:
Hi,

I think it would be good to include some way of defining acess priviledges (beyond acess type from SW side).

A pattern that always occurs in register map design, is the need to make "some registers accessible only in certain situations".

This typically occurs in:
- Some registers are targeted for manufacturing / DFT testability and you don't want them to be writable by user
- Some registers are security critical, and should be accessed only when machine is in certain state (e.g. life-cycle state of silicon, CPU machine mode, etc...)

A possible reactions when an access to "forbidden" register is executed could be:
- Ignore Write Access
- Read access returns all zero.
- Return error on the bus "slv_err". IMHO this could be optionally configurable at generation time.

Implementation-wise, I can imagine that there could be a pre-defined user defined property, e.g. something like "p_access_right", and value of this property would be name of the signal that would be used to gate the access for this block. Then it would be simple to lock acess to all registers by a single signal, or have multiple separate locks. This is usefull when there are more device states than just locked / unlocked.

amykyta3, 2024-05-01:
SystemRDL already provides a mechanism for this using the `swwe` and `swwel` properties. See: https://peakrdl-regblock.readthedocs.io/en/latest/props/field.html#swwe-swwel

If assigned a boolean, a per-field input is inferred. Alternatively a reference to a common RDL `signal` or `field` component can be assigned to control multiple.

Bulk assignments of this property can be easily done using the RDL default assignment keyword.

Blebowski, 2024-05-01:
Again, thanks, I did not know about this feature. Combined with the "signal" reference, this is awesome.

closed 2024-05-01 by Blebowski