SystemRDL/PeakRDL-regblock #111: External registers packed struct

benoitdenkinger, 2024-05-23:
Hello,

I'm using external register in SystemRDL. The regblock plugin generates packed structure for external registers. For example:

From SystemRDL description:

```
    reg {
        desc = "UART live status register";
        field {
            desc = "TX buffer is full";
            sw = r;
            hw = rw;
            we = true;
        } txfull[0:0] = 0x0;
        field {
            desc = "RX buffer is full";
            sw = r;
            hw = rw;
            we = true;
        } rxfull[1:1] = 0x0;
        field {
            desc = "TX FIFO is empty";
            sw = r;
            hw = rw;
            we = true;
        } txempty[2:2] = 0x1;
        field {
            desc = "TX FIFO is empty and all bits have been transmitted";
            sw = r;
            hw = rw;
            we = true;
        } txidle[3:3] = 0x1;
        field {
            desc = "RX is idle";
            sw = r;
            hw = rw;
            we = true;
        } rxidle[4:4] = 0x1;
        field {
            desc = "RX FIFO is empty";
            sw = r;
            hw = rw;
            we = true;
        } rxempty[5:5] = 0x1;
    } external status @ 0x10;
```

I get in the regblock package:

```
    typedef struct packed {
        logic txfull;
        logic rxfull;
        logic txempty;
        logic txidle;
        logic rxidle;
        logic rxempty;
        logic [25:0] _reserved_31_6;
    } uart_INTF_d878ad62__status__external__fields__in_t;
```

I have problem with this because I believe the bit order of the status register is not the one I expect. txfull is placed at the MSB in a packed array struct, but from my SystemRDL code I'm expecting it to be at the LSB. The generated code is also misleading as there is a reserved space _reserved_31_6 but it does not correspond to the 6 to 31 bit range when assigning the struct to a logic 32 bits variable.

Am I missing something or is there indeed a problem here?

benoitdenkinger, 2024-05-24:
If I revert the packed structure order I indeed get the behavior I expect:

```
    typedef struct packed {
        logic [25:0] _reserved_31_6;
        logic rxempty;
        logic rxidle;
        logic txidle;
        logic txempty;
        logic rxfull;
        logic txfull;
    } uart_INTF_d878ad62__status__external__fields__in_t;
```

I'll implement this in my fork, let me know if this aligns with your vision and I'll propose a PR.

amykyta3, 2024-12-20:
Thanks for pointing this out! Will be fixed in the upcoming release.

amykyta3, 2024-12-20:
Implemented in v0.23.0

closed 2024-12-20 by amykyta3