SystemRDL/PeakRDL-regblock #120: [BUG] intr field can miss events during SW writes

hughjackson, 2024-10-04:
- [x] I have reviewed this project's [contribution guidelines](https://github.com/SystemRDL/PeakRDL-regblock/blob/main/CONTRIBUTING.md)

**Describe the bug**
When a field is defined as intr, the next field can be used to set a bit (assuming the default sticky behaviour). In the current implementation the next value is ignored on a cycle when a SW write is happening to the register. The systemRDL spec seems to imply this behaviour in the general case, but it feels as though it is contrary to the interrupt case.

Details like these can be helpful:
```systemrdl
    reg {
        default hw = w;
        field {
            intr;
        } event = 0;
    } status;
```
Gives

   ```systemverilog
 // Field: rf.status.event
    always_comb begin
        automatic logic [0:0] next_c = field_storage.status.event_.value;
        automatic logic load_next_c = '0;
        if(decoded_reg_strb.status && decoded_req_is_wr) begin // SW write
            next_c = (field_storage.status.event_.value & ~decoded_wr_biten[0:0]) | (decoded_wr_data[0:0] & decoded_wr_biten[0:0]);
            load_next_c = '1;
        end else if(hwif_in.status.event_.next != '0) begin // stickybit
            next_c = field_storage.status.event_.value | hwif_in.status.event_.next;
            load_next_c = '1;
        end
        field_combo.status.event_.next = next_c;
        field_combo.status.event_.load_next = load_next_c;
    end
```

**Expected behavior**
Option 1: For a given field, bits set in next should be set unless they are being cleared by the write
Option 2: For a given field, bit set in next should be set unless at least 1 bit in the field is being cleared

Option1 and 2 are the same for single bit fields.

**Additional context**
Option 2 could be achieved by overriding get_predicate in WriteOneClear and sibling classes (untested code below):

```python
class WriteOneClear(_OnWrite):
    comment = "SW write 1 clear"
    onwritetype = OnWriteType.woclr

    def get_onwrite_rhs(self, reg: str, data: str, strb: str) -> str:
        return f"{reg} & ~({data} & {strb})"

    def get_predicate(self, field: 'FieldNode') -> str:
        # only treat as a write if one of the bits is high
        return f"{super.get_predicate(field)} && |({data} & {strb})"
```

I'm happy to work on a fix, but thought it wise to get input on whether a fix is appropriate, and which way to fix it.

hughjackson, 2024-12-07:
This would fix the single bit case. The multi-bit case still has some odd behaviour.

If the field is 2 bits with a starting value of 2'b01 and the HW is trying to set bit 0 and SW is using woclr to clear bit 1.
precedence=sw => the field will go to 2'b00
precedence=hw => the field will go to 2'b11
expected behaviour: this field will go to 2'b10 (each bit is handled individually)

This is likely a limitation of systemrdl itself. Perhaps some additional option to support this mode, such as precedence_per_bit

If I replaced my 2 bit field with 2 single bit fields, then precedence=hw will give the correct behaviour (since the HW controls are per field).

This doesn't feel like a problem for normal sw writes, as the software is writing all bits in the field in general (ignoring byte masks). This is a specific issue for woset and woclr where the software is updating only a specific subset of fields/bits

amykyta3, 2024-12-20:
I see what you mean. What you are describing is beyond sw/hw precedence, but rather a bitwise "bias" towards 1 or 0 during sw+hw collisions.
Indeed this is not something that is described by the SystemRDL language.

For interrupts, I would definitely recommend using individual single-bit fields. Not only does the `woclr`/`woset` mechanism work more cleanly, it till make your register spec more explicit and easier for the reader.

Closing since the original issue described is not a bug, and the multi-bit precedence behavior is out of scope.

closed 2024-12-20 by amykyta3