SystemRDL/PeakRDL-regblock #123: Regblock reset

philipaxer, 2024-11-04:
Hi, 

I have some general questions which to some degree are SystemRDL specific, but on the other hand have also some relevance to the compiler/generator.

### Question -  SystemRDL Example Clause 17.1
I went over the the example in the SystemRDL spec, Clause 17.1, the relevant snippted is below.

```
signal {
 name="PCI Soft Reset";
 desc="This signal is used to issue a soft reset to the PCI(E) device";
 activelow; // Define this signal is active low
 async; // define this reset type is asynchronous
 field_reset; // define this signal to reset the fields by default
 // This signal will be hooked to registers PCI defines as NOT Sticky.
 // This means they will be reset by this signal.
} pci_soft_reset;
 signal {
 name="PCI Hard Reset";
 desc="This signal the primary hard reset signal for the PCI(E) device";
 async; // define this reset type is asynchronous
 activelow; // Define this signal as active low
 cpuif_reset; // This signal will be or'd with the PCI Soft Reset Signal
 // to form the master hard reset which will reset all flops.
 // The soft reset signal above will not reset flops that PCI
 // defines as STICKY.
 } pci_hard_reset;
```

First of all, the interpretation of what a hard/soft reset is not well-defined. My assumption is this: soft-reset only clears registers (i.e. field storage elements in the bank). hard-reset will clear the registers/fields and reset all state towards the cpu_intf side (aka the entire IP block). But i have no idea what the editor of the spec had in mind. I had length discussions about hard/soft resets at my previous company :)

For `pci_soft_reset` PeakRDL will generate a logic input into the register block which is used as aync reset of the cpu_intf decoder FSM. So far so good.

The comment in the SystemRDL spec says this 
```/ This signal will be or'd with the PCI Soft Reset Signal
 // to form the master hard reset which will reset all flops.
 // The soft reset signal above will not reset flops that PCI
 // defines as STICKY.```

I dont understand that comment. There is no ORing, and I also don't see why there should be ORing as per the spec. Also there is nothing sticky in this example.

### Question - Self Clearing Reset bit
This may or may not be something for the SystemRDL regblock documenation. 

I want to design a bit that acts as a reset for the register bank. On my first attempt I created a RESET bit in a control register and just wired that into the rst input of the bank (without dealing with ANY othe the signal/cpuif_reset/field_reset magic whatsoever.) Naturally that doesnt work well. The reset will kill the AXI-lite FSM the AXI interface hangs. 
Then I created a signal similar to the pci_soft_reset as and feed the `hwif_out.CONTROL.RESET` into the newly created "soft" reset. That seems to work. It would be great to have a recommended way, in case different cpuif decoders act differently and potenially create different bahavior. 

Thanks 
Philip

amykyta3, 2024-11-29:
# Hard vs soft reset
Agree it is difficult to infer from the example what the author would have meant by "hard" or "soft" reset, as well as the comments. I'm inclined to assume this was simply a sloppily explained example by the spec authors and leave it at that. Also agree that "hard" and "soft" resets can have numerous meanings depending on context, and may not really be relevant to the spec.

# Self-clearing reset bit
There are a few ways to accomplish this, but my favorite is probably using a `singlepulse` field whose output drives a `field_rest` input signal.

```
addrmap my_regmap {
    signal {
        sync;
        activehigh;
        field_reset;
    } my_soft_reset;

    reg {
        field {
            sw=w;
            hw=r;
            singlepulse;
        } soft_reset = 0;
    } reset_reg;

    // ... Other regmap contents
};
```

What is happening:
* The creation of `my_soft_reset` signal will infer an auxiliary input to the register block:  `hwif_in.my_soft_reset`
* Its property `field_reset` instructs the compiler to use this as a reset signal for all fields hierarchically enclosed in the block it was instantiated in. In this case, anything inside `my_regmap`. This is equivalent to assigning each field the property `resetsignal = my_soft_reset;`.
* The `soft_reset` field is declared as a write-only `singlepulse` which creates a single-cycle pulse.

Technically this is still incomplete. What remains is an external connection in Verilog:
```
assign hwif_in.my_soft_reset = hwif_out.reset_reg.soft_reset.value;
```

The plus side of this technique is that it controls field resets of all fields based on *hierarchy* automatically. This can be expanded further where you can control multiple reset domains quite reliably.
Downside is that it is not fully self-contained and requires an external connection.

# Reset bit - Other method
Another way is to use more direct assignments of the `resetsignal` property:
```
addrmap my_regmap {
    reg {
        field {
            sw=w;
            hw=r;
            singlepulse;
        } soft_reset = 0;
    } reset_reg;

    default resetsignal = reset_reg.soft_reset;

    // ... Other regmap contents
};
```

This has the benefit of being fully self-enclosed, however there is one big caveat: Default property assignments only apply to *lexical scope*, not *hierarchical scope*.
This means that if you had a block or register declared outside of the `my_regmap` namespace, it will not inherit the reset.

If course you could directly assign the property everywhere, but that gets extremely tedious.

Hope that helps!

closed 2024-12-13 by amykyta3

amykyta3, 2026-10-05:
RDL doesn't have a great way to describe secondary resets. If the desired value of the alternate reset is all 0's or 1's, then you could use `hwclr` or `hwset` properties. These can be assigned an RDL signal that could act as the alternate reset input.