SystemRDL/PeakRDL-regblock #130: [BUG] External register struct order is flipped

stevobailey, 2024-12-30:
- [X] I have reviewed this project's [contribution guidelines](https://github.com/SystemRDL/PeakRDL-regblock/blob/main/CONTRIBUTING.md)

The order of fields in an external register struct is flipped. Example RDL:
```
addrmap csr {
    default accesswidth = 32; 
    default regwidth = 32; 
    addressing = regalign;

    external reg {
        field {
            hw = r;
            sw = w;
        } lsb[0:0] = 0x0;
    } REG @ 0x0;
};
```

Compile command:
```
peakrdl regblock example.rdl -o regblock --cpuif passthrough --default-reset arst_n
```

Output package file:
```
...
    typedef struct packed {
        logic lsb;
        logic [30:0] _reserved_31_1;
    } csr__REG__external__fields__out_t;

    typedef struct {
        logic req;
        logic req_is_wr;
        csr__REG__external__fields__out_t wr_data;
        csr__REG__external__fields__out_t wr_biten;
    } csr__REG__external__out_t;
...
```

The output Verilog just assigns the whole 32-bit write bus to the `wr_data` struct:
```
    assign hwif_out.REG.req = decoded_req_is_wr ? decoded_reg_strb.REG : '0; 
    assign hwif_out.REG.req_is_wr = decoded_req_is_wr;
    assign hwif_out.REG.wr_data = decoded_wr_data;
    assign hwif_out.REG.wr_biten = decoded_wr_biten;
```

But the `lsb` field appears first in the struct, which means the MSB of the `decoded_wr_data` gets assigned to the `lsb` field. This can be fixed by flipping the order of fields in the struct, or by assigning specific bits of `decoded_wr_data` to fields within the `wr_data` struct, as done with internal registers.

amykyta3, 2024-12-30:
Please make sure you are using the latest peakrdl-regblock. This was
recently fixed in v0.23.0:
https://github.com/SystemRDL/PeakRDL-regblock/releases/tag/v0.23.0
Issue #111

stevobailey, 2024-12-31:
Thanks, that did it.

closed 2024-12-31 by stevobailey