SystemRDL/PeakRDL-regblock #133: [BUG] Tool is not working

DonRum, 2025-01-15:
- [x] I have reviewed this project's [contribution guidelines](https://github.com/SystemRDL/PeakRDL-regblock/blob/main/CONTRIBUTING.md)

**Describe the bug**
I installed peakrdl using pip and attempted to generate an SV file for the atxmega_spi.rdl example using the command:
`peakrdl regblock atxmega_spi.rdl -o regblock/ --cpuif apb3-flat`
However, no output was produced. Interestingly, the program seems to function correctly, as it returns an error when no input file is provided.

* No error messages, just pause for like 3 second. 
* Python: 3.12, 3.13 (I tried both)
* OS:
```
Edition	Windows 11 Pro
Version	24H2
OS build	26100.2894
Experience	Windows Feature Experience Pack 1000.26100.36.0
```

**Expected behavior**
Generating of .sv file.

amykyta3, 2025-01-15:
That command will dump the output SV files to a folder named "regblock" can
you confirm whether that folder is being created?

Also since this is windows, try omitting the trailing "/" on that arg.

amykyta3, 2025-01-25:
closing due to inactivity

closed 2025-01-25 by amykyta3