SystemRDL/PeakRDL-regblock #137: swmod does not respect wstrb

bman12three4, 2025-03-30:
When writing to swmod fields with wstrb=0, the swmod output is still pulsed even though the data in the register has not been modified.

Looking at the generated code, it appears that it only cares about 2 things:
1. Did the incoming address match the field?
2. Was the incoming request a write?

It does not look at wr_biten to determine if the field has actually been written or not.

[image]

the backstory here is that I have an axi lite bus which has been converted from 128 bits down to 32 bits. When accessing a 32 bit subset, it still accesses the other 96 bits with wstrb=0. If these extraneous accesses happen to end up on a swmod field, swmod should not be pulsed since the user did not modify the fields.

In this particular case, it happens that it is the same swmod field that is accessed twice, but that is only because the registerfile is small and the address wraps around.

amykyta3, 2025-04-09:
Thanks for reporting!
Yeah agree that writes should also account for the biten to check if modified. Should be a pretty straightforward fix

amykyta3, 2025-04-09:
Some implementation notes to self:
* Move `sw_onwrite._OnWrite._wr_biten()` utility method (and others) up into the `FieldLogic` class
* Update `FieldLogic.get_swmod_identifier()` method to also use the above `get_wr_biten()` utility.

amykyta3, 2025-04-11:
Fixed! Will be published in next release.

closed 2025-04-11 by amykyta3