SystemRDL/PeakRDL-regblock #141: [BUG] RuntimeError: More than one exporter plugin was registered with the same name 'regblock'

galaviel, 2025-05-08:
Hi,

Using the below recipe to clone and install from github and then a simple regblock exporter command but getting the below. Any help greatly appreciated !

RuntimeError: More than one exporter plugin was registered with the same name 'regblock': 
	... temp_dir/github/PeakRDL-regblock/src/peakrdl_regblock/__peakrdl__.py:Exporter
	... temp_dir/github/PeakRDL-regblock/src/peakrdl_regblock/__peakrdl__.py:Exporter

My recipe: 
############
## CLONE   #
############
    mkdir -p github
    /usr/m/pkgs/git/2.18.0/bin/git clone https://github.com/SystemRDL/systemrdl-compiler.git     github/systemrdl-compiler
    /usr/m/pkgs/git/2.18.0/bin/git clone https://github.com/SystemRDL/PeakRDL-regblock.git       github/PeakRDL-regblock
    /usr/m/pkgs/git/2.18.0/bin/git clone https://github.com/SystemRDL/PeakRDL.git                github/PeakRDL
############
## VENV    #
############

# create python virtual env
/usr/m/pkgs/python/3.10.6/bin/python3.10 -m venv venv_github

# active the virtual env
source  ./venv_github/bin/activate.csh
############
## PIP     #
############
./venv_github/bin/pip install -e github/systemrdl-compiler/
./venv_github/bin/pip install -e github/PeakRDL-regblock/  
./venv_github/bin/pip install -e github/PeakRDL/peakrdl-cli/

-------------------------------------

My runtime command: 
./venv_github/bin/peakrdl regblock cmdq.rdl -o regblock_cmdq/ --cpuif apb3-flat 

Thanks in advance, Gal.

galaviel, 2025-05-08:
updates:
1. I noticed an improvement after upgrading PIP; apparently PIP supports editable installs from pyprojet.toml only since version [21.3] - see  https://stackoverflow.com/questions/69711606/how-to-install-a-package-using-pip-in-editable-mode-with-pyproject-toml
2. after running venv/bin/pip install on the above x3 clones, I removed (rm -rf) all the *.egg-info directories I could find in those clones.
3. This seemed to work - the regblock is running now!

amykyta3, 2025-05-08:
Yeah older versions of `pip` are weird about editable installs. You can get into a situation where you have two copies of a package installed simultaneously - one "normal" install, and a second editable install.

To avoid confusing precedence issues, the peakrdl CLI will intentionally throw a runtime error if it detects two plugins of the same name. In this case, it is being thrown off by the two copies of the same install.

Sounds like you got it working. Let me know if there is anything actionable, or if this issue is ok to close.
Thanks for using PeakRDL!

galaviel, 2025-05-11:
Thanks allot for your support and the additional information. Indeed it's working now.
I'm new to Python packaging so the conversation is very helpful.
Perhaps consider requiring a more advanced version of pip as a prerequisite for install ? Although I can see why not to require this - only developers want this, most people won't need this.
Thanks again - appreciated!
Gal.

closed 2025-05-11 by galaviel