SystemRDL/PeakRDL-regblock #143: AttributeError: module 'peakrdl_regblock.cpuif' has no attribute 'axi4lite'

Tooniis, 2025-05-28:
I'm trying to follow https://peakrdl-regblock.readthedocs.io/en/latest/cpuif/customizing.html to modify the AXI4-Lite interface to match the one used in [PULP AXI](https://github.com/pulp-platform/axi), but I'm getting `AttributeError: module 'peakrdl_regblock.cpuif' has no attribute 'axi4lite'`. `dir(peakrdl_regblock.cpuif)` returns this:
```
['CpuifBase', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', 'apb4', 'base']
```
Any idea why `axi4lite` (and many others apparently) are missing?

I have `peakrdl-regblock` 1.0.0 installed with pip.

amykyta3, 2025-05-30:
Hard for me to tell what is going on - Could you provide more details?

- Show the full traceback of the error. Not just the last error. This has useful context.
- Provide the contents of the class that implements axi4lite variant from PULP
- How are you loading the new Cpuif class? Using the config TOML? If so, please show contents
- What is the command-line invocation you are using?

Tooniis, 2025-06-04:
I'm using it in a SCons build script but its reproducible as a regular standalone python script:

```
import peakrdl_regblock
import re

class PULP_AXI_LITE(peakrdl_regblock.cpuif.axi4lite.AXI4Lite_Cpuif):
        @property
        def port_declaration(self) -> str:
                return "AXI_LITE.Slave s_axil"

        def signal(self, name:str) -> str:
                signal_name = name.lower()
                return "s_axil." + re.sub('^(aw|ar|b|w|r)', '\1_', signal_name)
```

```
Traceback (most recent call last):
  File "<...>.py", line 4, in <module>
    class PULP_AXI_LITE(peakrdl_regblock.cpuif.axi4lite.AXI4Lite_Cpuif):
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'peakrdl_regblock.cpuif' has no attribute 'axi4lite'
```

I'm trying to invoke regblock from Python (not using command-line tool) as shown in the wiki, using `RegblockExporter`.

amykyta3, 2025-06-04:
I see now. This is somewhat intentional to prevent excessive module imports within the tool.
Please change your implementation to import the class directly:
```python
from peakrdl_regblock.cpuif.axi4lite import AXI4Lite_Cpuif
```