SystemRDL/PeakRDL-regblock #151: [BUG] Not Reset Condition for External Register Assertions

lizhiliTT, 2025-07-11:
Hello Team PeakRDL,

I am using Synopsys VCS. The external register assertions are violated due to X's at t = 0 ns. May a reset condition be added to suppress this?

Also, it would be nice if the assertions were named.

Thank you!

[image]

amykyta3, 2025-07-16:
Agreed! Will adjust

amykyta3, 2025-07-16:
Looking at the implementation, I am already masking the assertion with the reset.
The assertion you are observing is actually due to how your testbench drives the `clk` and `rst_n` signals at T=0.

I assume in your case, these are declared uninitialized and then are procedurally assigned a value. This results in a "rising edge" transition of the clock: `X->1` at T=0. Since rst_n is also X at this moment, it is treated as a non-reset condition which triggers the assertion.

This is a pretty common issue in SystemVerilog testbenches. The easiest way to avoid this is to use initial assignments for `clk` and `rst_n`. These will clear the erroneous X at T=0:

```verilog
logic rst_n = '0;
logic clk = '0;
initial forever begin
    #5ns;
    clk = ~clk;
end
```

You'll also notice that in the above code, I initialize `clk` to 0, and defer the first rising-edge to happen later. This can also help avoid assertion noise at T=0.

amykyta3, 2025-07-16:
Added names to assertions for the next release. Let me know if there is anything else needed related to this topic. Otherwise will close as completed.

closed 2025-07-16 by amykyta3