SystemRDL/PeakRDL-regblock #155: [FEATURE] Add option to change readback implementation

rskvprd, 2025-07-20:
*Problem*
I had timing and resource utilisation issues with large register map. Significant part of the logic utilisation and timing issues was in readback logic.

*Solution*
But when i changed implementation from `Flat 1-hot array then OR reduce`[^1] to `Flat case statement`[^1] (in fact, i've generated big `if..else..else..else` statement), utilisation was significantly reduced and timing issue was fixed.

*Proposal*
It would be nice to have option like `--readback-mux-impl` with options like `orreduce` and `case` for designs with low budget/large regmaps

[^1]: See [readback-mux devnotes](https://github.com/SystemRDL/PeakRDL-regblock/blob/main/docs/dev_notes/template-layers/5-readback-mux)

rskvprd, 2025-10-08:
It was Xilinx FPGA

amykyta3, 2025-12-12:
Implemented. Will be in next release

closed 2025-12-12 by amykyta3

Blebowski, 2025-12-24:
Hi @amykyta3 ,

no, I did not make the experiment to confirm. The write-up above comes from 5-6 years back when I was examining the difference between OR tree vs case. Maybe DC now also has the special "if-else" pass. 

I prefer to write RTL that:
- "more closely resembles HW" (e.g. don't rely on operand sharing, code time-shared arithmetics explicitly)
- "inuition tells me this can give better result" (e.g. the `if-else` infers chain that needs to be later parallelized / optimized, rather than case that infers parallel logic right up front).

even if compiler is able to optimize it well. Maybe I am too conservative since I was taught design by people who used to do it in schematics in the old days...

That being said: Talk is cheap, show me the "numbers" in this case :)
When I get back to work, I can run an example in DC and provide some numbers.

davekeeshan, 2025-12-24:
> Well @davekeeshan you are right to some extent. But if I have 20 peripherals with hundreds of registers and, thousands of flip-flops, those 4 percents in each peripheral do add-up to maybe interesting area being used that would not need to be...

Yes I absolutely acknowledge that, and for some designs that is a number one requirement. I am coming from a design reuse perspective, which is more concerned with faster turnaround times, you are more likely to sacrifice area for engineer time. 

Again if area is a major concern, one should probably be hand optimising every block. We typically don't do that any more, and it it is code bloat and area we see that. 

Wearing my fpga / asic hat, having code that has been extensively used on an fpga gives you serious warm fuzzies when you are dropping it into an asic. More than 4%

amykyta3, 2026-03-19:
I was able to kick the tires on this a bit on several different synthesis environments. Performance seems comparable to initial experiments. Will release this change shortly.
Thanks all for the analysis and comments.